contiki-ng/contiki-ng #988: motes leave the network and rejoin frequently

prodomo, 2019-06-25:
Hi, 
I use examples/rpl-udp/udp-{server,cilent}.c and turn on tsch with 32 motes(using hardware CC1350), but motes(client) would join the network and leave frequently.
motes joined the network few mins and leave, and would like to rejoin again.
What should I do to improve this situation?

debug-ito, 2019-06-27:
I suspect TSCH time synchronization failure. Have you examined TSCH log? It can be enabled by writing `#define LOG_CONF_LEVEL_MAC LOG_LEVEL_DBG` in your project-conf.h.

prodomo, 2019-06-27:
[2019/05/28, 12:30:42] [ERR : TSCH Queue] ! add packet failed: 0 #0x20001880 0 #0x0 #0x0
[2019/05/28, 12:30:42] [ERR : TSCH      ] ! can't send packet to 0012.4b00.0d2e.6784 with seqno 3, queue 0 4

I saw this error log many times, I changed TSCH Queue size but it didn't work.
What should I do?

debug-ito, 2019-06-28:
That log means that the transmit queue is full and so a frame is discarded. I think that's probably because the node doesn't get ACK from the destination node, so it keeps to retransmit the same frame, so it doesn't consume the transmit queue fast enough.

There could be several reasons why the node doesn't get ACK. Maybe it doesn't transmit radio signal in the first place. Maybe there is strong radio interference. Maybe the time synchronization fails between the nodes, so either the DATA frame or ACK frame were missed. I would increase the log level (or maybe insert some debug code) to determine which is the main cause.

closed 2019-10-04 by g-oikonomou