contiki-ng/contiki-ng #994: Scheduling of ASF bursty links possibly breaks hard cells

mbaddeley, 2019-06-28:
So in tsch-slot-operation.c, burst_link_requested and burst_link_scheduled allow nodes to opportunistically schedule another TX if they have more packets in the TX queue (as per ASF, I believe).

```
/* We requested an extra slot and got an ack. This means the extra slot will be scheduled at the received */
    if(burst_link_requested) {
        burst_link_scheduled = 1;
    }
```

I was playing around with statically defined schedules, and I noticed that when I explicitly set cells as a hard RX, a node still might try to schedule a bursty TX, causing collisions with other TXs from other nodes in the schedule. e.g. the schedule should be...

```
1 - Rx | Rx | Rx
2 - Tx | Rx | Rx
3 - Rx | Tx | Rx
4 - Rx | Rx | Tx
```

... but node 2 has additional packets in its TX queue, it schedules a burst slot, and this happens...

```
1 - Rx | Rx  | Rx
2 - Tx | Tx* | Rx
3 - Rx | Tx* | Rx
4 - Rx | Rx  | Tx

*collision
```

Is this is a bug, or am I just not getting something?

atiselsts, 2019-06-28:
My understanding is that this is the expected behavior as per the IEEE 802.15.4 standard. 

With the MSF and other scheduling functions that adapt to the traffic, enabling the burst bit reduces performance. With purely autonomous scheduling the discussion is more open - for some applications the burst bit helps.

yatch, 2019-06-28:
I don't think it's a bug. It seems kind of a policy, which prioritizes transmission over reception. IEEE 802.15.4-2015 doesn't say anything about how the transmitter should behave in this particular case (see Section 7.2.1.3), although the pending bit itself is underspecified....

However, as you mentioned, bursty TX may break a communication schedule. Busty TX may be undesirable in some cases. In my working project, I've introduced a configuration macro with which a user (developer) can decide to use the bursty TX or not. I will propose it soon.

mbaddeley, 2019-06-28:
Sounds good! I think it would be good to have this in NG at some point so it's on by default but if a user wants to break from IEEE 802.15.4 then they can, as it would be useful when making your own scheduling functions. In the meantime, like @yatch, I'll define my own macro :)

yatch, 2019-06-28:
@mbaddeley Try this one. Set `TSCH_CONF_PENDING_BIT_ENABLED` to `0` in your `project-config.h`: https://github.com/contiki-ng/contiki-ng/pull/995 It's not tested yet, though. :-(  I will add a simple test there.

yatch, 2019-06-28:
This is a solution to disable the bursty TX which is already available, suggested in @atiselsts [comment](https://github.com/contiki-ng/contiki-ng/pull/995#issuecomment-506680688)

> setting TSCH_BURST_MAX_LEN to 0 or 1.

simonduq, 2019-06-29:
I believe the standard leaves completely open the policy on Tx side (when to set bit pending or not) but mandates the behavior on the Rx side (listen at next slot if bit was set).

Something nice to have would be configurable policies on the Tx side. For instance, we could choose never to set the bit if there's a cell about to run next slot. Or even device to only use the pending bit for neighbors we have no dedicated cells to. A configurable callback "should_set_bit_pending" maybe?

atiselsts, 2020-09-13:
The new IEEE 802.15.4-2020 standard is more clear in this point:

> When operating in TSCH mode, the frame pending bit can only be set if there is not a link scheduled in the following timeslot

This "link scheduled" check  is clearly meant to happen on the transmitter's side. Of course, the transmitter may not be aware of the links scheduled in the receiver. The TSCH schedule can still be broken by the frame pending bit.

alexrayne, 2020-10-14:
> Of course, the transmitter may not be aware of the links scheduled in the receiver. The TSCH schedule can still be broken by the frame pending bit.

if transmiter not aware receiver scheduling - it can produce collisions besides pending bits. So, check scheduling on a transmiter side is good solution.

If we don`t look ahead for the next slot schedule on transmiter side, and use break pending transmition by loose slot in arbitrage - then we pay for this on receiver side by unexpected receives packets, that should be droped by adress filtering. This is some energy penalty.

atiselsts, 2020-10-20:
The default behavior is fixed in #1408.

closed 2020-10-20 by atiselsts