contiki-ng/contiki-ng #999: Heapmem does not return correctly aligned addresses

akindestam, 2019-07-05:
In heapmem.c the ```heap_base``` buffer is allocated as an array of ```char```:
```
static char heap_base[HEAPMEM_ARENA_SIZE];
```
 Since ```char``` has an alignment of only 1 byte the compiler is free to start this array at any byte (for reproductions sake: I am building on ARM with GCC and -Os). As heapmem seems to align addresses against the start of this buffer, you can see how this gives bad results when you have HEAPMEM_ALIGNMENT set to 4.

A solution would be to define using ```__attribute__((aligned(HEAPMEM_ALIGNMENT)))``` at least on compilers which support this GCC extension:
```
__attribute__((aligned(HEAPMEM_ALIGNMENT)))
static char heap_base[HEAPMEM_ARENA_SIZE];
```

Allocating the array out of a type with a suitable alignment might be another potentially more portable way, although making that configurable might be tricky. Example for HEAPMEM_ALIGNMENT == 4:
```
```

atiselsts, 2019-07-05:
There's `CC_ALIGN` macro that could be used instead of explicitly invoking GCC attributes.

akindestam, 2019-07-05:
> There's `CC_ALIGN` macro that could be used instead of explicitly invoking GCC attributes.

Ah, thanks for pointing that out!

g-oikonomou, 2019-08-31:
Fixed in #1001

closed 2019-08-31 by g-oikonomou