contiki-ng/contiki-ng #1008: CCS/CC1310: Rtimer issue with simplelink platform

chinacoolge, 2019-07-11:
I found an issue that caused exception when using rtimer in contiki-ng. 

**When "TARGET = simplelink" is configured in the makefile , the MCU is down at the rtimer_callback().
When changed  "TARGET = cc26x0-cc13x0" ,everyting is ok.**

_Here is my test program._
////////////////////////////////////////////////////////////////////////

```
#include <stdio.h>
#include "contiki.h"
#include "sys/etimer.h"
#include "sys/stimer.h"
#include "sys/timer.h"
#include "sys/rtimer.h"

PROCESS(timer_process, "ETimer x Timer x STimer Process");
AUTOSTART_PROCESSES(&timer_process);

static int counter_etimer;
static int counter_timer;
static int counter_stimer;
static struct timer timer_timer;
static struct stimer timer_stimer;
static struct ctimer timer_ctimer;
static struct rtimer timer_rtimer;

/*---------------------------------------------------------------------------*/
void
ctimer_callback(void *ptr)
{
  /* rearm the ctimer */
  ctimer_reset(&timer_ctimer);
  printf("CTimer callback called\n");
}
/*---------------------------------------------------------------------------*/
void
rtimer_callback(struct rtimer *timer, void *ptr)
{
  /* Normally avoid printing from rtimer - rather do a process poll */

  //printf("RTIMER_NOW(): %d\n",(int)RTIMER_NOW());
  printf("RTimer callback called\n");
  /* Re-arm rtimer */
  rtimer_set(&timer_rtimer, RTIMER_NOW() + RTIMER_SECOND / 2, 0,
             rtimer_callback, NULL);
}
/*---------------------------------------------------------------------------*/
/*
 * This Process will count timer expires on one e-timer (that drives the
 * events to the process), one timer, and one stimer.
 *
 */
PROCESS_THREAD(timer_process, ev, data)
{
  static struct etimer timer_etimer;

  PROCESS_BEGIN();

  ctimer_set(&timer_ctimer, CLOCK_SECOND, ctimer_callback, NULL);
  printf("RTIMER_NOW(): %d\n",(int)RTIMER_NOW());
  rtimer_set(&timer_rtimer, RTIMER_NOW() + RTIMER_SECOND / 2, 0,
             rtimer_callback, NULL);

  while(1) {
    /* set all the timers */
    timer_set(&timer_timer, 3 * CLOCK_SECOND);
    stimer_set(&timer_stimer, 3);
    etimer_set(&timer_etimer, 3 * CLOCK_SECOND);

    PROCESS_WAIT_EVENT_UNTIL(ev == PROCESS_EVENT_TIMER);

    counter_etimer++;
    if(timer_expired(&timer_timer)) {
      counter_timer++;
    }

    if(stimer_expired(&timer_stimer)) {
      counter_stimer++;
    }

    printf("Timer process: %s\n", counter_timer == counter_etimer
           && counter_timer == counter_stimer ? "SUCCESS" : "FAIL");
  }

  PROCESS_END();
}
/*---------------------------------------------------------------------------*/
```

hseokman, 2019-07-11:
cc26x0-cc13x0 and simplelink are considered different platform, so they have little different implementation of rtimer.

arurke, 2024-07-06:
hi @chinacoolge did you try removing the `printf` in `rtimer_callback`? Simplelink-platform does not like printf in interrupt-context (I believe it is mentioned in the doc. somewhere)

closed 2025-04-30 by arurke