contiki-ng/contiki-ng #1009: How can I test NDP in cooja

wspark91, 2019-07-14:
HI!!
I want to know how to test 6lowpan-neighbor discovery without using RPL in a cooja environment.

Thank you...

yatch, 2019-07-16:
Although I'm not sure what you want to do, you may want to check `UIP_CONF_ND6_` settings in `os/net/ipv6/uip-nd6.h` or `os/contiki-default-conf.h`. To my understanding, [RFC 6775](https://tools.ietf.org/html/rfc6775) and its accompanying spec are not supported yet.

closed 2019-07-16 by yatch

wspark91, 2019-07-17:
I will try the test as shown below.
[image]

wspark91, 2019-07-17:
> Although I'm not sure what you want to do, you may want to check `UIP_CONF_ND6_` settings in `os/net/ipv6/uip-nd6.h` or `os/contiki-default-conf.h`. To my understanding, [RFC 6775](https://tools.ietf.org/html/rfc6775) and its accompanying spec are not supported yet.

I will try the test as shown below.
[image]