contiki-ng/contiki-ng #1012: contiki-ng over the air download

0608rohit, 2019-07-17:
I am trying to perform Over The Air Download on simplelink LAUNCHXL-CC1352R1 using contiki-ng
If it supports OAD then is it possible to do multi broadcasting OAD? 

How am I suppose to start with OAD and multi broadcasting OAD using contiki-ng ? (Documentation and Link)

0608rohit, 2019-07-19:
Hello,
i am expecting answer from community.

yatch, 2019-07-19:
It's not supported yet, although there is [an on-going work on OAP/OTA](https://github.com/contiki-ng/contiki-ng/pull/623).

By the way, Gitter would be a better place to ask this kind of question.  See https://github.com/contiki-ng/contiki-ng/wiki/Where-to-report-issues,-ask-questions,-etc.%3F#gitter

closed 2019-07-19 by yatch