contiki-ng/contiki-ng #1014: cc26xx web-demo for LAUNCHXL-CC1352R1

0608rohit, 2019-07-22:
Hello, 
I can't  build cc26xx web-demo code for LAUNCHXL-CC1352R1 in Contiki-ng.
 please tell me which types of changes are needed for CC1352R1 ?

0608rohit, 2019-07-24:
Hello
I want to build example/platform-specific/cc26xx/cc26xx-web-demo code on cc13x2 platform.
is it possible? if yes then,
What step should I have to follow to change platform-specific example?(document would be helpful for me)
I am expecting answer from you.

g-oikonomou, 2019-08-31:
This particular example is not intended for use with the simplelink platform. It has not been tested against it at all, so even if you could make it compile I would be surprised if you didn't encounter other issues. I can immediately imagine problems with the example's BLE advertisements as well as with the way it collects various sensor readings.

closed 2019-08-31 by g-oikonomou