contiki-ng/contiki-ng #1026: Probing stops after global repair.

pciolkosz, 2019-08-15:
The problem is this line and the next one:
https://github.com/contiki-ng/contiki-ng/blob/7b4de0585224d877de53d1402df28d4d31f3eef9/os/net/routing/rpl-lite/rpl-dag.c#L231

While other timers gets rescheduled, the probing timer is not scheduled in init_dag_from_dio function, but in process_dio_init_dag, which is not called in case of global repair. This causes the node to stop probing. This sometimes leads to a group of nodes joining the new version, but not being able to find any preferred parent in it, because of high etx to the DIO sender.

simonduq, 2019-08-24:
ood catch! Maybe we should call `process_dio_init_dag` instead.

g-oikonomou, 2019-10-25:
Fixed in #1077

PR #1077 (2019-10-24): Misc. RPL-Lite fixes
Fixes #1033
Fixes #1026 

And also re-works how the preferred parent is cleared upon removing all neighbors.
closed 2019-10-25 by g-oikonomou