contiki-ng/contiki-ng #1033: Global repair does not remove default route.

pciolkosz, 2019-08-23:
Hi,
I believe there is a bug in the global_repair_non_root function. It calls init_dag_from_dio to reinitialize the DAG, which calls init_dag, which does:
https://github.com/contiki-ng/contiki-ng/blob/7b4de0585224d877de53d1402df28d4d31f3eef9/os/net/routing/rpl-lite/rpl-dag.c#L483
This sets the preferred parent to NULL without calling rpl_neighbor_set_preferred_parent, which removes and adds default routes to preferred parent. This may lead to a node with no preferred parent sending packets to it's previous preferred parent. If a node has a preferred parent and performs global repair, this preferred parent is left as a default route. If after the repair, this node is left with no parent, it will use this leftover route as the default one.

simonduq, 2019-08-24:
I agree, I think we should be calling `rpl_neighbor_set_preferred_parent` before the memset. Feel free to submit a fix as PR!

simonduq, 2019-08-24:
Good catch! Maybe we should call `process_dio_init_dag` instead.

g-oikonomou, 2019-10-25:
Fixed in #1077

PR #1077 (2019-10-24): Misc. RPL-Lite fixes
Fixes #1033
Fixes #1026 

And also re-works how the preferred parent is cleared upon removing all neighbors.
closed 2019-10-25 by g-oikonomou