contiki-ng/contiki-ng #1038: Add TSCH support to the SimpleLink platform

g-oikonomou, 2019-08-31:
The SimpleLink platform's radio drivers do not support all the features required by TSCH. This is proving to be a popular platform and some users have been bitten by trying to use TSCH.

I have changed the platform's wiki page to document this lack of support.

This is something we want to be adding, so a roadmap item here.

https://github.com/contiki-ng/contiki-ng/wiki/Platform-simplelink

Yagoor, 2019-10-29:
Are you aware if Texas Instrument is working on this, since they are the maintainers of this platform port? 
If not I can work on it because I have the physical board at home.

Yagoor, 2019-11-04:
@g-oikonomou

Mano2Code, 2019-12-03:
> Are you aware if Texas Instrument is working on this, since they are the maintainers of this platform port?
> If not I can work on it because I have the physical board at home.

@Yagoor  Do you have a solution for this?

jeandudey, 2019-12-11:
What needs to be done specificly to get this working?

hseokman, 2019-12-11:
I have somewhat working version implemented for CC1312R
https://github.com/hseokman/contiki-ng/tree/tsch-cc1312r
Please provide feedback.
I had to change rtimer interrupt priority so that the radio API work in the interrupt handler and RTC and RAT was drifting away that I had to sync periodically. Also printing in the interrupt handler will reset the device that I had to disable framer and MAC logging.

stainlessinoxx, 2020-04-25:
@hseokman thanks! I’m trying your code on a CC1352. So far I’ve got 5-25% packet transfer success rate, using a scope there’s about a 10 factor missing in your sending time.

Someone also needs to point me to the TSCH algorithm documentation. I’ll get it working, it’s a good start. Understanding exactly what’s missing in the “gluing” layer and addressing it will be a pleasure for me, since I’m paid to do it.

hseokman, 2020-04-27:
I tried to match CC1310 implementation, and here are what I did to make CC1312R to work. It's been running pretty good. CC1312R is sub Ghz radio, so I don't know what may not work in 2.4Ghz simplelink platform.

1. Added missing cc13xx-50kbps-tsch.c so that it uses the 50kbps timing
2. Match radio parameters to CC1310 in cc13xx-cc26xx-def.h
3. Lowed RTC interrupt priority so that radio command can be executed in the interrupt handler
4. Changed preamble bytes and added timestamp in the radio setup to match CC1310
5. Change runCmd to scheduleCmd in netstack_sched so that it does not hang. runCmd cannot run in the interrupt handler
6. Check and sync RAT to RTC when the radio is turned off and turned on. This supposed to work without manually doing it, but I found it was not. RAT and RTC kept drift off that device cannot get time synchronized after some time.

atiselsts, 2020-04-27:
@hseokman Sounds promising!

Regarding the 6th point. In `cc26xx-cc13xx/` we had to add a line to the initialization code to keeps the RTC and RAT synced even if the system goes to a low-power mode. From `clock.c`:

    /* enable sync with radio timer */
    HWREGBITW(AON_RTC_BASE + AON_RTC_O_CTL, AON_RTC_CTL_RTC_UPD_EN_BITN) = 1;

hseokman, 2020-04-27:
@atiselsts 
Thanks for pointing out the code.
However it did not work for me. RAT and RTC was still getting out of sync.

stainlessinoxx, 2020-04-28:
No luck on my side either. I've generated a joint (ans synchronized) runtime log with both the coord and the node, accessible [here](https://1drv.ms/u/s!AtTXz5gzhH25vX8WryWhI7jU3uRK?e=IBELIh).
We see the coordinator somewhat detecting the node, but failing to establish a valid route. The node then apparently dies... I also joined my project-conf.h and main routine for the reference...

hseokman, 2020-04-28:
@stainlessinoxx It seems you have the same problem that I tried to fix with #6 above. Your device receives an EB, but it fails any communication after that, so it left the network. When the device receives an EB, it starts synchronize with the coordinator and wake up radio when necessary. You can try setting some pin high to see how they are synchronized. Have some pin go high at the start of the slot operation and go low at the end. You can search for TSCH_DEBUG_SLOT_START and TSCH_DEBUG_SLOT_END. Then connect an oscilloscope and measure how they are different.

atiselsts, 2020-04-30:
@hseokman Thanks a lot for your effort. I started looking at the code and testing it using the CC2652R1 devices that I have. I was able to hack together working IEEE mode code that has very reliable packet reception.

The RTC and RAT are syncing just fine (for a while), as long as the RAT is periodically restarted with its start/stop commands, there is no need for manual compensation,

A couple more issues that took longer to solve:
1) the `receiving_packet` / `pending_packet` do not behave in the way TSCH is expecting them to behave. This was also very difficult to get right in CC26xx-CC13xx, so no surprise. I was not able to fix `receiving_packet` / `pending_packet` logic for IEEE mode yet, so I made a minor change in the TSCH driver instead. 
2) the function `cmd_rx_cb` is called and the `rf_sched_process` polled even when the radio is in poll mode. This may not be critical, but is very confusing. I disabled the `rf_sched_process` polling.

A minor issue is that `RAT_TIMESTAMP_OFFSET` needed adjustment. The existing values in the driver are simply copied over from CC26xx-CC13xx platform code, by the look of it.

Remaining issues: currently the code works for around 12 minutes until the RAT/RTC suddenly breaks completely. Could be related to RAT overflow handing.

atiselsts, 2020-04-30:
BTW I can't test the prop mode as I don't have CC13x2, and the simplelink platform does not work at all for me on CC1310 launchpads - is this supposed to happen?

hseokman, 2020-04-30:
@atiselsts I only tested with CC1312 and I kind of merged simplelink IEEE mode and cc26x0-cc13x0 prop mode to make this new prop mode.
1. I took the code for receiving_packet/pending_packet from cc26x0-cc13x0 and put in prop mode.
2. I will have to change that.
3. I measured RAT_TIMESTAMP_OFFSET with oscilloscope, but this could be different for different radio.

closed 2020-07-03 by g-oikonomou

minithc, 2022-02-21:
@atiselsts
> Remaining issues: currently the code works for around 12 minutes until the RAT/RTC suddenly breaks completely. Could be related to RAT overflow handing.

Same issue with endless loop "Unable to restore RX command" / "Unable to schedule FS command" after a while (5 minutes - several days).
Clearing the queue in that situation helps:
RF_flushCmd(&rf_netstack, RF_CMDHANDLE_FLUSH_ALL, RF_ABORT_PREEMPTION);