contiki-ng/contiki-ng #1040: doxygen fail inside the docker image

g-oikonomou, 2019-08-31:
Following a post by @jeppenodgaard on gitter, I looked into doxygen inside docker (macOS host).

Self explanatory:

```
$ contiker bash
user@b7a031bfb111:~/contiki-ng$ cd tools/doxygen/
user@b7a031bfb111:~/contiki-ng/tools/doxygen$ doxygen
error: file `Doxyfile' not found
error: Could not create output directory /home/user/contiki-ng/tools/doxygen/html
Exiting...
user@b7a031bfb111:~/contiki-ng/tools/doxygen$ doxygen Doxyfile
error: file `Doxyfile' not found
error: Could not create output directory /home/user/contiki-ng/tools/doxygen/html
Exiting...
user@b7a031bfb111:~/contiki-ng/tools/doxygen$ ls Doxyfile 
Doxyfile
user@b7a031bfb111:~/contiki-ng/tools/doxygen$ head -n1 Doxyfile 
# Doxyfile 1.8.8
user@b7a031bfb111:~/contiki-ng/tools/doxygen$ make
> Cleaning Documentation
 done.
error: file `Doxyfile' not found
error: Could not create output directory /home/user/contiki-ng/tools/doxygen/latex
Exiting...
Makefile:27: recipe for target 'html' failed
make: *** [html] Error 1
user@b7a031bfb111:~/contiki-ng/tools/doxygen$ 
```

So I am in the right directory, `Doxyfile` does exist and the user can read it, but doxygen fails for whatever reason. This causes a cascade of other errors: i) `tools/readthedocs` will fail, ii) `tests/00-documentation`.

I am trying to figure out the cause but so far no luck.

g-oikonomou, 2019-10-12:
pointing doxygen to a Doxyfile that resides anywhere outside the mounted macOS dir removes the problem... (but of course the documents won't build properly since doxygen cannot find the sources).

I am suspecting this to be a doxygen thing, not a file system thing.

@yatch can you confirm? Any clues?

yatch, 2019-10-15:
It works in my environment...

```
$ contiker bash
user@db1fa299153d:~/contiki-ng$ cd tools/doxygen
user@db1fa299153d:~/contiki-ng/tools/doxygen$ make
Searching for include files...
Searching for example files...
Searching for files in directory /home/user/contiki-ng/tools/doxygen
Searching for files in directory /home/user/contiki-ng/examples/hello-world
Searching for images...
Searching for dot files...
Searching for msc files...
Searching for dia files...
Searching for files to exclude
Searching INPUT for files to process...
...
Generating alphabetical compound index...
Generating hierarchical class index...
Generating member index...
Generating file index...
Generating file member index...
Generating example index...
finalizing index lists...
writing tag file...
lookup cache used 18788/65536 hits=233224 misses=19348
finished...
```

Here is an example running directly with docker options:
```
$ docker run --mount type=bind,source=/Users/yatch/Work/contiki-ng,destination=/home/user/contiki-ng -ti contiker/contiki-ng bash
user@8a9d9ab72a56:~/contiki-ng$ cd tools/doxygen/
user@8a9d9ab72a56:~/contiki-ng/tools/doxygen$ doxygen
Searching for include files...
Searching for example files...
```
What do you get with this docker command?

The version info of my environment:
```
$ git rev-parse --abbrev-ref HEAD
release-4.4

$ git rev-parse HEAD
605dd5e5d1061828d10da83becead140b9384261

$ docker images --digests contiker/contiki-ng
REPOSITORY            TAG                 DIGEST                                                                    IMAGE ID            CREATED             SIZE
contiker/contiki-ng   latest              sha256:09f573a5412939a4c34b18cfc3b56ef53475c099291d5c0d6604f58063bb5e08   64c5549b6d6b        2 weeks ago         3.49GB

$ docker version
Client: Docker Engine - Community
 Version:           19.03.2
 API version:       1.40
 Go version:        go1.12.8
 Git commit:        6a30dfc
 Built:             Thu Aug 29 05:26:49 2019
 OS/Arch:           darwin/amd64
 Experimental:      false

Server: Docker Engine - Community
 Engine:
  Version:          19.03.2
  API version:      1.40 (minimum version 1.12)
  Go version:       go1.12.8
  Git commit:       6a30dfc
  Built:            Thu Aug 29 05:32:21 2019
  OS/Arch:          linux/amd64
  Experimental:     true
 containerd:
  Version:          v1.2.6
  GitCommit:        894b81a4b802e4eb2a91d1ce216b8817763c29fb
 runc:
  Version:          1.0.0-rc8
  GitCommit:        425e105d5a03fabd737a126ad93d62a9eeede87f
 docker-init:
  Version:          0.18.0
  GitCommit:        fec3683
```

g-oikonomou, 2019-10-15:
How interesting...

```
$ docker run --mount type=bind,source=/Users/.../contiki-ng,destination=/home/user/contiki-ng -ti contiker/contiki-ng bash
user@595ca46b1ac5:~/contiki-ng$ cd tools/doxygen/
user@595ca46b1ac5:~/contiki-ng/tools/doxygen$ doxygen Doxyfile
error: file `Doxyfile' not found
error: Could not create output directory /home/user/contiki-ng/tools/doxygen/html
Exiting...
```

```bash
$ git show
commit 605dd5e5d1061828d10da83becead140b9384261
Merge: 737ff74 4ef0d57
Author: George Oikonomou <george@contiki-ng.org>
Date:   Fri Oct 11 15:30:11 2019 +0100

    Merge pull request #1023 from yatch/pr/docker-sysctl
    
    docker: fix an issue on net.ipv6.conf.all.disable_ipv6

$ docker images --digests contiker/contiki-ng
REPOSITORY            TAG                 DIGEST                                                                    IMAGE ID            CREATED             SIZE
contiker/contiki-ng   latest              sha256:09f573a5412939a4c34b18cfc3b56ef53475c099291d5c0d6604f58063bb5e08   64c5549b6d6b        2 weeks ago         3.49GB
contiker/contiki-ng   <none>              sha256:1f914c2b451e3de7858b0180975b7cf37584b296e18d6963e3ccf928c370d4b8   8307892c9c77        11 months ago       3.32GB
contiker/contiki-ng   <none>              sha256:924219fd641715cc7f3607125a64ace69d26e25fb781d5046b2607ebbafc0238   604f13cd4689        16 months ago       3.33GB

$ docker version
Client: Docker Engine - Community
 Version:           19.03.2
 API version:       1.40
 Go version:        go1.12.8
 Git commit:        6a30dfc
 Built:             Thu Aug 29 05:26:49 2019
 OS/Arch:           darwin/amd64
 Experimental:      false

Server: Docker Engine - Community
 Engine:
  Version:          19.03.2
  API version:      1.40 (minimum version 1.12)
  Go version:       go1.12.8
  Git commit:       6a30dfc
  Built:            Thu Aug 29 05:32:21 2019
  OS/Arch:          linux/amd64
  Experimental:     true
 containerd:
  Version:          v1.2.6
  GitCommit:        894b81a4b802e4eb2a91d1ce216b8817763c29fb
 runc:
  Version:          1.0.0-rc8
  GitCommit:        425e105d5a03fabd737a126ad93d62a9eeede87f
 docker-init:
  Version:          0.18.0
  GitCommit:        fec3683
```

yatch, 2019-10-15:
It is interesting... I don't have my Mac now; I will see further later.

g-oikonomou, 2019-10-15:
I'll try to test on a different mac and report back

g-oikonomou, 2019-10-15:
Different mac, same problem... (This one is Mojave, the other one Catalinea so that's not it).

Then I grabbed a fresh clone on the host and and changed the mount point:

```
$ docker run --mount type=bind,source=/Users/.../cng,destination=/home/user/cng -ti contiker/contiki-ng bash
user@d8bbaf43b6dc:~/contiki-ng$ cd ../cng 
user@d8bbaf43b6dc:~/cng$ cd tools/doxygen/
user@d8bbaf43b6dc:~/cng/tools/doxygen$ doxygen Doxyfile 
error: file `Doxyfile' not found
QDir::currentDirPath: stat(".") failed
QDir::currentDirPath: stat(".") failed
QDir::currentDirPath: stat(".") failed
QDir::currentDirPath: stat(".") failed
QDir::currentDirPath: stat(".") failed
error: Failed to open temporary file /doxygen_objdb_8.tmp
Exiting...
```

yatch, 2019-11-27:
Now I have the same problem on my another Docker environment, which is built on Windows PC using WSL:

```
user@12e5c8289885:~/contiki-ng/tools/doxygen$ make
> Cleaning Documentation
 done.
error: file `Doxyfile' not found
QDir::currentDirPath: stat(".") failed
QDir::currentDirPath: stat(".") failed
QDir::currentDirPath: stat(".") failed
QDir::currentDirPath: stat(".") failed
QDir::currentDirPath: stat(".") failed
error: Failed to open temporary file /doxygen_objdb_1245.tmp
Exiting...
Makefile:27: recipe for target 'html' failed
make: *** [html] Error 1
```

I spent some time trying to find a solution yesterday, but unfortunately it's not resolved yet.