contiki-ng/contiki-ng #1046: Cooja build broken with Java 11

romain-jacob, 2019-09-02:
Hello there,

I am setting up a fresh install, and notices that building Cooja requires 'java.xml.bind', which (apparently) has been deprecated since JAVA 9 and now completely removed in JAVA 11:
https://stackoverflow.com/questions/52502189/java-11-package-javax-xml-bind-does-not-exist

One can always downgrade to JAVA 10, but that is not great. It would be important to find a proper fix imo. Suggestions welcome.

Cheers,
-- 
Romain

mbaddeley, 2019-09-02:
I think this is part of the bigger 32-bit issue. As well as supporting later java versions, I think it really needs to be ported to 64-bit machines so it doesn't need to be run on Docker.

https://stackoverflow.com/questions/52502189/java-11-package-javax-xml-bind-does-not-exist

romain-jacob, 2019-09-02:
Hum, I do not know whether this has anything to do with the 32- vs 64-bit... One can "fix" the issue by installing java 8 (`sudo apt-get install java-8-openjdk`), then execute the `update-alternative --config java` to select that java8 should be used...

Works, but that's not a great fix I think.

iancook1701, 2019-12-15:
I have tried installing java 8, and all I get is "E: Unable to locate package java-8-openjdk"

Any thoughts on this? Has anyone gotten it working?

alexandruioanp, 2019-12-16:
Maybe you need to add the PPA (https://www.geofis.org/en/install/install-on-linux/install-openjdk-8-on-ubuntu-trusty/).

iancook1701, 2019-12-18:
That did fix the issue I was having, thanks!

arurke, 2022-05-16:
This seems to have been resolved by https://github.com/contiki-ng/cooja/pull/21

closed 2022-05-16 by arurke