contiki-ng/contiki-ng #1054: Neighbor table consistency

atiselsts, 2019-09-03:
In the system, there are a number of structures that neighbor tables:
- RPL candidate parents
- link stats
- IPv6 neighbors (called `ds6_neighbors`)
- TSCH neighbors (not an actual `NBR_TABLE`, but used in a similar role)
- other tables - to be reviewed!

**The problem:** There is no attempts to ensure that the tables contain the same entries.

This can (and does) lead to issues, for example, as the TSCH neighbors and link stats are removed much more agressively than RPL candidate parents.
- If there is an entry for RPL parent, but a TSCH neighbor cannot be added for this parent, the parent cannot be probed.
- If the entry for link stats is deleted for a RPL parent, it can be re-added back later with a better metric. This causes the routing tree convergence time to become very long.

**Proposed solution**: introduce a global consistency mechanism between the tables.

Details:
- Could use an approach with bitmaps to implement that: only remove a neighbor when all of TSCH, RPL and link stats have freed that neighbor, i.e. the bitmap is zero.
- Common neighbor replacement policy can now be implemented, e.g. "least recently used".
- Additionally, there should be a locking / prioritization option (as there is already?). For example, a TSCH timesource or RPL preferred parent should never be removed to free space for a new neighbor that is not one of these.
- Allow to disable the IPv6 neighbor table completely, to save RAM. As discussed with George, in the normal Contiki-NG use case there is no need to keep that table at all, because the MAC address / link address mapping can be done on the fly.

simonduq, 2019-09-25:
Not denying that there are issues, but "There is no attempt [..]" is a false statement :p
The very purpose of `nbr-table` was to attempt to sync all tables; for context see https://github.com/contiki-os/contiki/pull/288

Now I agree that TSCH being not a `nbr-table` bring back inter-table inconsistencies.

Also, note that there is already in place both a bitmap to keep track of all tables in the system, a lock and a replacement policy, as well as `UIP_CONF_ND6_AUTOFILL_NBR_CACHE` to operate without IPv6-MAC mapping... but given the frustrations expressed here I assume NG and new expectations have outgrown this good old `nbr-table` :p 😢

simonduq, 2019-09-25:
Big fail if we still have inconsistencies between RPL and link-stats, given my 2013 statement from the PR linked above "The goal is to make all these modules maintain information on a consistent set of neighbors, and avoid e.g. to have a RPL preferred parent but no link metric for it." :p

atiselsts, 2019-09-26:
Ok, I stand corrected - clearly the current neighbor table mechanism is aiming to add the consistency to neighbors.

Given this, perhaps it better to change the way how the current neighbor tables are used rather than introduce new code.

Changing TSCH neighbors to a neighbor table would be one the things to do.

Introducing a clear hierarchy could be another. Perhaps we can say for each RPL candidate parent, we always need both a link stats entry and a TSCH neighbor. If this is not possible (the tables are full with locked entries), new RPL candidate parents are not added.

Currently, link stats are "reset" by removing all link stats neighbor. That could be changed to re-initializing the link stats neigbhors in a fresh state rather than deleting them.

Getting rid of the IPv6 neigbhors (optionally turned on, with a `#define`) would be nice optimization.

simonduq, 2019-09-26:
Right. So a current limitation of `nbr-table` is that it isn't interrupt-safe, and this is why TSCH doesn't use it. Would be great to find a way to fix this. High value but not sure how much effort.

> Introducing a clear hierarchy could be another. Perhaps we can say for each RPL candidate parent, we always need both a link stats entry and a TSCH neighbor. If this is not possible (the tables are full with locked entries), new RPL candidate parents are not added.

The situation today is already that whenever a `nbr-table` inserts a new entry, spaces is always reserved for all other tables in the system. You get a new entry with a bitmap all set to 0 ready to become 1 whenever any module needs it. So if you succeed to add neighbor X in RPL you have a guarantee you'll also get an entry for link-stats or any other module. It's also already the case that if all entries are locked then no entry can be added. See `NBR_TABLE_FIND_REMOVABLE` for custom replacement policy.

atiselsts, 2020-04-16:
@simonduq I'm starting to take a look at the TSCH neighbor code.
One thing I noticed is that TSCH neighbors are kept in linked list, which is also not interrupt safe - that's why you have the `tsch_is_locked()` guards, right?

So on the face of it, it's not clear why the list could not be replaced with a Contiki-NG neighbor table. Any thoughts?

simonduq, 2020-04-16:
Right, the neighbor list doest indeed require locking (the packet queue it contains it what supports atomic r/w).

Also, I re-checked the assumptions and they are as follows:
* TSCH interrupts do not write to anything accessible outside of interrupt context unless it has atomic read/write
* The lock is only used by non-interrupts to read/write to non-atomic shared structures
* Lock is taken for write but only simply checked for read (as nothing will interrupt the read to take the lock)
* TSCH interrupts are skipped if lock is taken/requested (though we only seem to check `tsch_lock_requested` not `tsch_locked`... bug?

So I think switching to neighbor table in principle could work, with a caveat: the neighbor table has built-in GC that might kick in anytime another module adds a new neighbor, and might remove a TSCH neighbor without taking the TSCH lock.

I believe this can be worked around using the neighbor table entry lock, which forbids GC (but we need to think about a solution that doesn't keep all TSCH neighbors lock at all times... maybe `tsch_pending_events_process` could do something, like take the TSCH lock and then allow neighbor table to run GC somehow?)

atiselsts, 2020-04-16:
TSCH neighbors that don't have packets and dedicated slots in the schedule are automatically removed in `tsch_queue_free_unused_neighbors()`. Maybe that's enough? I mean, we could keep all neigbor table entries with a TSCH neighbor locked in the neighbor table, as you say. When the TSCH neighbor is removed by the `tsch_queue_free_unused_neighbors()` function, the neighbor table entries become GC'able.

simonduq, 2020-04-16:
Yes, I think this should work.

atiselsts, 2020-04-16:
Ok, then I'll look into creating a PR that replaces the TSCH neighbors with
a neighbor table.

closed 2020-05-13 by atiselsts

alexrayne, 2020-05-22:
[@atiselstsl](https://github.com/atiselstsl): can you explain about `nbr_table_keys` - what purpose of this list? 
Why table of lladdr`eses cant be just plain table, whiout list? This listing takes about twice memory waste, and what effort it provide?

PR #1239 (2020-05-13): Move TSCH neighbors to a neighbor table
Reduces memory usage and partially fixes #1054 .