contiki-ng/contiki-ng #1056: CC1352P and CC1350 compatibility

maxgonch, 2019-09-05:
I setup modules with a default radio settings 802.15.4 channel: 24
Frequency is equals to 867,925 MHz (0x0363.0xECCC for RF_cmdFs)
But CC1352P and CC1350 does't hear each other.
Analizing radio with SDR dongle showed that CC1352 with contiki-ng transmit packets on a frequency with  the offset 60kHz.
And so if I changed for CC1352P frequency command RF_cmdFs to 0x0363.0xFC32 modules became visible for each other.

TI samples works fine with default settings 0x0363.0xECCC. No frequency shift occur.
Why contiki-ng create 60 kHz offset on CC1352P?

tiepettersen, 2019-09-06:
Are the load capacitors on the 48 MHz crystal mounted on the CC1352P board? If so, then you need to disable the internal capacitor array. This can be done by applying the lowest modifying value to the capacitor array by setting the following defines in your project settings:

```c
#define SET_CCFG_MODE_CONF_XOSC_CAP_MOD         0x0
#define SET_CCFG_MODE_CONF_XOSC_CAPARRAY_DELTA  0xD5
```

maxgonch, 2019-09-06:
Great thanks! It works fine.

closed 2019-09-06 by maxgonch