contiki-ng/contiki-ng #1061: Valgrind errors in tcp-socket.c

alexandruioanp, 2019-09-10:
Steps to reproduce
```
> contiker
$ cd examples/mqtt-client/
$ make TARGET=native
$ sudo valgrind -v ./mqtt-client.native
```
Snipped output:
```
==5407== HEAP SUMMARY:
==5407==     in use at exit: 0 bytes in 0 blocks
==5407==   total heap usage: 1 allocs, 1 frees, 1,024 bytes allocated
==5407== 
==5407== All heap blocks were freed -- no leaks are possible
==5407== 
==5407== ERROR SUMMARY: 10 errors from 4 contexts (suppressed: 0 from 0)
==5407== 
==5407== 1 errors in context 1 of 4:
==5407== Source and destination overlap in memcpy(0x806f534, 0x806f534, 123)
==5407==    at 0x4030D39: memcpy (in /usr/lib/valgrind/vgpreload_memcheck-x86-linux.so)
==5407==    by 0x8056FF2: tcp_socket_send (tcp-socket.c:353)
==5407==    by 0x804F74C: send_out_buffer (mqtt.c:274)
==5407==    by 0x80510ED: publish_pt (mqtt.c:680)
==5407==    by 0x8052731: process_thread_mqtt_process (mqtt.c:1258)
==5407==    by 0x8053ED7: call_process (process.c:190)
==5407==    by 0x80540BB: do_event (process.c:296)
==5407==    by 0x80540DC: process_run (process.c:310)
==5407==    by 0x8060377: platform_main_loop (platform.c:303)
==5407==    by 0x8053056: main (contiki-main.c:164)
==5407== 
==5407== 
==5407== 1 errors in context 2 of 4:
==5407== Source and destination overlap in memcpy(0x806f534, 0x806f534, 2)
==5407==    at 0x4030D39: memcpy (in /usr/lib/valgrind/vgpreload_memcheck-x86-linux.so)
==5407==    by 0x8056FF2: tcp_socket_send (tcp-socket.c:353)
==5407==    by 0x804F74C: send_out_buffer (mqtt.c:274)
==5407==    by 0x8051284: pingreq_pt (mqtt.c:729)
==5407==    by 0x8052319: process_thread_mqtt_process (mqtt.c:1219)
==5407==    by 0x8053ED7: call_process (process.c:190)
==5407==    by 0x80540BB: do_event (process.c:296)
==5407==    by 0x80540DC: process_run (process.c:310)
==5407==    by 0x8060377: platform_main_loop (platform.c:303)
==5407==    by 0x8053056: main (contiki-main.c:164)
==5407== 
==5407== 
==5407== 1 errors in context 3 of 4:
==5407== Source and destination overlap in memcpy(0x806f534, 0x806f534, 53)
==5407==    at 0x4030D39: memcpy (in /usr/lib/valgrind/vgpreload_memcheck-x86-linux.so)
==5407==    by 0x8056FF2: tcp_socket_send (tcp-socket.c:353)
==5407==    by 0x804F74C: send_out_buffer (mqtt.c:274)
==5407==    by 0x805034C: connect_pt (mqtt.c:454)
==5407==    by 0x8052008: process_thread_mqtt_process (mqtt.c:1186)
==5407==    by 0x8053ED7: call_process (process.c:190)
==5407==    by 0x80540BB: do_event (process.c:296)
==5407==    by 0x80540DC: process_run (process.c:310)
==5407==    by 0x8060377: platform_main_loop (platform.c:303)
==5407==    by 0x8053056: main (contiki-main.c:164)
==5407== 
==5407== 
==5407== 7 errors in context 4 of 4:
==5407== Source and destination overlap in memcpy(0x806f534, 0x806f554, 480)
==5407==    at 0x4030D39: memcpy (in /usr/lib/valgrind/vgpreload_memcheck-x86-linux.so)
==5407==    by 0x8056835: acked (tcp-socket.c:79)
==5407==    by 0x8056B96: appcall (tcp-socket.c:205)
==5407==    by 0x8056D30: process_thread_tcp_socket_process (tcp-socket.c:244)
==5407==    by 0x8053ED7: call_process (process.c:190)
==5407==    by 0x80541C9: process_post_synch (process.c:366)
==5407==    by 0x805862B: tcpip_uipcall (tcpip.c:804)
==5407==    by 0x805AD0E: uip_process (uip6.c:2107)
==5407==    by 0x8057D89: packet_input (tcpip.c:183)
==5407==    by 0x80581A6: eventhandler (tcpip.c:439)
==5407==    by 0x80586CC: process_thread_tcpip_process (tcpip.c:832)
==5407==    by 0x8053ED7: call_process (process.c:190)
==5407== 
==5407== ERROR SUMMARY: 10 errors from 4 contexts (suppressed: 0 from 0)
```
Compiling and running this on Fedora 30 x64 (gcc 9.2.1 with WERROR=0 to get around some incompatibilities) results in 0 Valgrind errors.

g-oikonomou, 2019-09-10:
Will take a little time to dig into this. I wonder whether this may also be the reason why I was seeing #724

alexandruioanp, 2019-09-10:
I can confirm this also happens when using the Vagrant image.

PR #1062 (2019-09-15): Replace memcpy in tcp-socket
This commit fixes #1061.

`memcpy()` should not be used if the source and destination memory locations alias.

In `acked()` the expectation is clearly that the two buffers overlap.

In `tcp_socket_send()` it's a bit trickier. This issue was discovered because the MQTT library was using a single memory location for 2 things: the buffer that it uses to construct its output packet and the buffer that it passes to the TCP socket for the output data. The TCP API does not require that the 2 buffers that are passed to `tcp_socket_send()` are non-overlapping and I think th [... cut]
closed 2019-09-15 by atiselsts