contiki-ng/contiki-ng #1063: cooja coap-example.The server does not respond to the client

chanseng, 2019-09-16:
I follow this tutorial to simulate coap's sever with client：
https://slogix.in/how-to-run-coap-applications-in-contiki-cooja-simulator
But after startup, the client will always report
--toggle timer--

--done--
It looks like server does not seem to respond to request.
what should i do.

chanseng, 2019-09-19:
the nodes output is below:
00:00.510 ID:2 Rime started with address 0.18.116.2.0.2.2.2
00:00.519 ID:2 MAC 00:12:74:02:00:02:02:02 Contiki 3.0 started. Node id is set to 2.
00:00.528 ID:2 nullsec nullmac nullrdc, channel check rate 128 Hz, radio channel 26, CCA threshold -45
00:00.538 ID:2 Tentative link-local IPv6 address fe80:0000:0000:0000:0212:7402:0002:0202
00:00.541 ID:2 Starting 'Erbium Example Server'
00:00.656 ID:1 Rime started with address 0.18.116.1.0.1.1.1
00:00.664 ID:1 MAC 00:12:74:01:00:01:01:01 Contiki 3.0 started. Node id is set to 1.
00:00.673 ID:1 nullsec CSMA ContikiMAC, channel check rate 8 Hz, radio channel 26, CCA threshold -45
00:00.685 ID:1 Tentative link-local IPv6 address fe80:0000:0000:0000:0212:7401:0001:0101
00:00.688 ID:1 Starting 'Border router process' 'Web server'
00:00.706 ID:1 ?PGot configuration message of type P
00:00.708 ID:1 Setting prefix aaaa::
00:01.173 ID:3 Rime started with address 0.18.116.3.0.3.3.3
00:01.182 ID:3 MAC 00:12:74:03:00:03:03:03 Contiki 3.0 started. Node id is set to 3.
00:01.191 ID:3 nullsec nullmac nullrdc, channel check rate 128 Hz, radio channel 26, CCA threshold -45
00:01.201 ID:3 Tentative link-local IPv6 address fe80:0000:0000:0000:0212:7403:0003:0303
00:01.204 ID:3 Starting 'Erbium Example Client'
00:01.207 ID:3 Press a button to request .well-known/core
00:01.690 ID:1 Server IPv6 addresses:
00:01.693 ID:1 aaaa::212:7401:1:101
00:01.696 ID:1 fe80::212:7401:1:101
00:11.199 ID:3 --Toggle timer--
01:17.064 ID:3
01:17.065 ID:3 --Done--
01:17.066 ID:3 --Toggle timer--
01:20.467 ID:3 |
01:20.468 ID:3 --Done--
01:20.469 ID:3 --Toggle timer--
01:20.489 ID:3 |
01:20.490 ID:3 --Done--
....toggle timer forever
someone please help me

kaleemullah360, 2019-09-20:
The tutorial you mentioned in link doesn't seems to be working.
my suggestion are you can use following example at mote side.

[https://github.com/contiki-ng/contiki-ng/blob/develop/examples/coap/coap-example-server/coap-example-server.c](https://github.com/contiki-ng/contiki-ng/blob/develop/examples/coap/coap-example-server/coap-example-server.c)

and following nodejs library at client (laptop/computer) side.
[https://github.com/mcollina/node-coap](https://github.com/mcollina/node-coap)

Here is instructions:

in cooja simulator add first mote as border router and from this mote menu choose 

**serial socket(SERVER).** and press start.

add second mote in simulation as **coap-example-server.c** mentioned above.

now setup node-coap library as global with following command:

`sudo npm i -g node-coap`

look for example provided with library or you can subscribe to a coap url of the form 
[coap://test/hello](coap://test/hello) via command line just to see if coap mote is sending its data over command line.

closed 2019-10-04 by g-oikonomou

chanseng, 2019-10-06:
thank you very much