contiki-ng/contiki-ng #1064: CSMA Link-layer security and rpl-border-router

tdesmet, 2019-09-17:
# Problem description

I am trying to get a network up with csma and link-layer security on the cc1310 and am facing some problems on getting it to work with the rpl border router.

The nodes are no problem, I use sensniff and wireshark and can see the secured packets and wireshark is able to decrypt them.

The problem is the rpl border router. It seems that something goes wrong. When I set LLSEC802154_CONF_ENABLED wireshark sees a lot of mallformed packets.

Either I am doing wrong or there is a bug in the code, but since there is no documentation on link-layer security I have no clue.

# What I did:
In the project-conf.h of the rpl-border-router and slip-radio example I put 
```c
#define LLSEC802154_CONF_ENABLED 1
#define CSMA_CONF_LLSEC_DEFAULT_KEY0 {0x9c, 0x96, 0x98, 0x6d, 0x30, 0xbd, 0x64, 0x86, 0x07, 0x8c, 0x39, 0x9a, 0xa1, 0x37, 0x44, 0xf7 }
```

# Wiresharh capture
I added the .txt extension so I could upload it, remove it after downloading to get the pcapng file.

[ll-security.pcapng.txt](https://github.com/contiki-ng/contiki-ng/files/3620431/ll-security.pcapng.txt)

NimRavi, 2019-09-19:
How to enable encryption & encrypt data in Contiki ng in AES 128. Does tsch-security provide it. If so how can it be done?

MoonPole, 2019-09-19:
How to encrypt data in Contiki ng in AES 128. Does tsch-security provide it? If so how can it be done?