contiki-ng/contiki-ng #1069: CoAP client on docker

g-oikonomou, 2019-09-21:
In our current docker image we use version 0.5.1 of the `coap-cli` package:

```
$ coap --version
0.5.1
$ coap get coap://[fd00::302:304:506:708]/.well-known/core
(2.05)	</.well-known/core>;ct=40,</test/hello>;title="Hello world: ?len=0..";rt="Text",</debug/mirror>;title="Returns your decoded message";rt="Debug",</test/chunks>;title="Blockwise demo";rt="Data",</test/separate>;title="Separate demo",</test/push>;title="Periodic demo";obs,</test/sub>;title="Sub-resource demo",</test/b1sepb2>;title="Block1 + Separate + Block2 demo"
```
Newer versions (0.6.0 or newer) will however fail

```
$ coap --version
0.7.0
$ coap get coap://[fd00::302:304:506:708]/.well-known/core
Invalid URL. Protocol is not given or URL is malformed.
```

(When trying natively on macOS I am getting a different error - a DNS resolve failure. Our theory is that something is messed up in newer versions with URL parsing)

This means that we are currently stuck: If a pull touches `Dockerfile` then travis will re-generate the container and the "bad" version of coap-cli will get installed. This is exactly what causes #1060 to fail (the pull touches `Dockerfile` to add vagrant tests).

@alexandruioanp

g-oikonomou, 2019-09-21:
In #1060 873c0ef2d24d40bcfa8e2cc2b366ab6d751ae630 therefore the proposal is to pin the coap client package to the known working version until this has been resolved.

Yagoor, 2019-09-30:
I took a look into this issue and I believe it shouldn't be classified as a bug in Contiki-NG.

The issue happens in this section of the coap-cli code:
[image]

By removing the try catch to see the real error, I got this:
[image]

Therefore this is outside the Contiki-NG scope. My guess is that out node.js is outdated and the new version of the coap-cli requires the latest version or a newer one at least.

Yagoor, 2019-09-30:
My suggestion would be to use a more stable version of the client. By stable I mean not using the node. This is one example.

http://manpages.ubuntu.com/manpages/bionic/man5/coap-client.5.html

I can work on updating the test, if you think it is necessary.

alexandruioanp, 2019-10-01:
As a note, I got the libcoap example client, which should be similar to the one you linked, to talk to the native Contiki CoAP server, so that could be a viable alternative. However, I think it's not available as a package in the version of Ubuntu that the Docker image uses so we might have to compile it from source.

g-oikonomou, 2019-10-04:
@Yagoor @alexandruioanp we are preparing for the upcoming release and it'd be nice to have something in that fixes this. Personal preference package instead of compile from source, if at all possible

Yagoor, 2019-10-06:
@alexandruioanp is correct. Libcoap is not available as a package for Ubuntu 16.04. Therefore the only solution would be to compile it from the source. Or to continue using the cli-coap with the version fixed.

g-oikonomou, 2019-10-18:
Switch to libcoap client (#1093) to be merged after the release. Closing this one.

closed 2019-10-18 by g-oikonomou