contiki-ng/contiki-ng #1089: cc1310 radio stops listening in busy network

tdesmet, 2019-10-11:
I have a network with 60 cc1310. Yesterday there was a moment with a lot of network traffic and after this moment I see a lot of nodes that are not reachable. Sniffing with sensniff I see the nodes sending out DIS packets and other nodes responding. But the affected nodes act like they do not hear the responses.

I have been debugging and noticed that the affected nodes don't seem to receive any packet anymore.

This seems related to this PR https://github.com/contiki-ng/contiki-ng/pull/567

The code base we are running on has this pr merged into it so there seems to be another issue where the radio stops listening for some reason.

g-oikonomou, 2019-10-11:
CSMA or TSCH?

tdesmet, 2019-10-11:
CSMA, cc26x0-cc13x0

g-oikonomou, 2019-10-11:
So under CSMA, there is a condition whereby the driver turns the radio off after TX. This is irrelevant for TSCH. Hence my question. If we have a different problem in the RF driver we need of course to look at it.

Also, #197 :)

tdesmet, 2019-10-11:
Can you point me to this condition to see if it is related? Thanks in advance

g-oikonomou, 2019-10-11:
#778, contiki-os/contiki#2161 and possibly others

tdesmet, 2019-10-11:
Yes I read those but cannot see how they are related to only csma :-)

Anyway I though that #567 was a fix for those.

g-oikonomou, 2019-10-11:
> Yes I read those but cannot see how they are related to only csma :-)

I think the fact that TSCH powercycles the radio anyway is what makes it disappear. I could be wrong of course, I do not understand exactly what the root cause is. Or whether they are related in the first place, just that they have similar manifestation

tdesmet, 2019-10-11:
Ok thank you, I am going to investigate further

I wonder if the simplelink platform has the same problems. Any clue?

g-oikonomou, 2019-10-11:
I do not know. But if we can create an example that breaks cc13x0-cc26x0 then we can try on the same h/w using simplelink. If the latter doesn't crash then added confidence that we need to be looking at cc26x0-cc13x0 prop mode driver.

tdesmet, 2019-10-11:
Ok I enabled debug in the radio driver and now see these messages

```
RXQ was full, re-enabling radio!
rf_cmd_prop_rx: send_cmd ret=0, CMDSTA=0x00000c23, status=0x0000
```

As expected the RX queue was full and we try to reenable the radio, but here it goes wrong. the call to rf_core_send_cmd in rf_cmd_prop_rx returns RF_CORE_CMD_ERROR.

Is there some documentation on the status codes returned? 0x0c23 is returned but what does it mean?

g-oikonomou, 2019-10-11:
THose codes are documented in the Technical Reference Manual, chapter 23 "Radio". This CMDSTA value is confusing me though

tdesmet, 2019-10-11:
Ok so I am not the only one not finding this value :-) 

Anyway what I see is that after this the radio remains off except when a packet is being send.

If I disable the transmit function from turning the radio off after transmitting then I can again receive packets untill it happens again.... so in periods I can receive packets.

tdesmet, 2019-10-14:
I tested the simplelink platform and I could not replay it. So it seams that simplelink is not affected.

MartinNPN, 2020-01-14:
Hello,

maybe there is an simple fix, try to replace the return value of rf_is_on function with: 
`return ( (smartrf_settings_cmd_prop_rx_adv.status == RF_CORE_RADIO_OP_STATUS_ACTIVE) || (smartrf_settings_cmd_prop_rx_adv.status == RF_CORE_RADIO_OP_STATUS_PROP_ERROR_RXBUF ) );`

closed 2023-01-13 by arurke