contiki-ng/contiki-ng #1092: Two RPL instances using one Bourder router in Cooja

philok93, 2019-10-12:
Hello,
I want to simulate two separate rpl instances in the same network. For example, I want 2 nodes to form a different rpl instance (lets say "A") and other 2 nodes to participate in rpl instance "B". These two rpl instances should have a common root/BR so that I can take metrics from both networks. Is that possible using rpl-lite? 

I want to do that so nodes participating in "A" instance will not participate in instance "B" (create two different networks but both to use same root).

Thanks a lot if anyone knows the answer.

yatch, 2019-10-15:
It seems `rpl-lite` doesn't support multiple instance:
https://github.com/contiki-ng/contiki-ng/wiki/Documentation:-RPL#rpl-lite
> RPL Lite removes support for storing mode in favor of non-storing mode, and removes the complexity of handling multiple instances and DODAGs. Through these changes, RPL Lite typically exhibits better performance and has a considerably smaller ROM footprint.

philok93, 2019-10-21:
Thank you @yatch for the information. I need to see how to implement my idea then in different way.

closed 2020-01-09 by philok93