contiki-ng/contiki-ng #1097: Contiki on custom board using CC2640F128 4x4mm package

asmakhalil, 2019-10-15:
Hello,

Due to tirtos not being supported on Linux, I have been trying to get contiki to work on my cc2640f128 on my custom board. Specifically, the cc26x0-demo example from the platform-specific code: https://github.com/contiki-ng/contiki-ng/blob/develop/examples/platform-specific/cc26x0-cc13x0/cc26x0-demo.c.
Now I have been running into a few problems but most importantly the fact that i can see my chip boot-up and yet not advertise, the changes i made where as follows:

- Used the srf06 platform with the cc2640 board as a base and started removing any button/led hal related code in the platform.c file since i am using my custom board (which to start is only powering my chip, no buttons or leds are being used)

- Altered the board.h file since the contiki one defines it for a 7x7mm packageand therefore used https://github.com/contiki-os/contiki/issues/2114 this post to guide me on the changes i need to make.

- Turned off the rom bootloader just as has been done on tirtos in order to eliminate any problems going wrong over there.

I am not sure why my chip is not advertising, I have hooked it up and downloaded tirtos stack and simple periphiral app in windows and got it to work, yet I have not been successful on linux/contiki, can anyone offer any help/direction on what to try? I am running out of ideas!

Thanks guys!

Asma

asmakhalil, 2019-10-15:
I should Add my chip is the 4x4mm or rsm package

g-oikonomou, 2019-10-15:
Hi

First reaction: check this function here, you may need to add a block for the CC2640 (I am not sure what the correct value is, but you should be able to find it easily).

Let us know how it goes

https://github.com/contiki-ng/contiki-ng/blob/develop/arch/cpu/cc26x0-cc13x0/rf-core/rf-core.c#L410

asmakhalil, 2019-10-15:
Thanks for your reply! will try this tomorrow and let you know! :)

g-oikonomou, 2019-10-16:
Fire up a pull request if you want. I would be surprised if you managed to achieve connected mode :)

closed 2019-10-16 by g-oikonomou