contiki-ng/contiki-ng #1099: Contiki-ng: CC26xx ble connected mode support

asmakhalil, 2019-10-17:
Hello!

After some research it appears as though Contiki does not support achieving connected mode in cc26xx chips. I am currently investigating the amount of work that would go into implementing this to see if it is a viable option for me to do with Contiki. I have already found some skeleton for connected mode in Contiki-ng code-base, so it looks promising with the support of the Contiki community. If anyone has any information on the scope of work that this would take/ has attempted it or something similar perhaps on a different TI CC chip your feedback would be very much appreciated! And of course any development on this from my end that achieves connected mode will go back to benefit the Contiki community.

Thanks!

Asma

asmakhalil, 2019-10-17:
skeleton code* no skeletons were found lol

asmakhalil, 2019-10-18:
Thanks for your reply!

We would like to investigate whether the TI drivers used by contiki(cc26xxware) support ATT/GATT and Contiki just needs a HAL implementation, or whether the TI drivers need an ATT implementation. 

If the latter, do you know how much effort would go into that?

Thanks!

Asma

spoerk, 2019-11-05:
As far as I know, the TI drivers do not support any ATT/GATT communication, but rather provide support for all low-level BLE radio primitives.

Unfortunately, I have no estimation on how long implementing such a support should take.