contiki-ng/contiki-ng #1101: CC1310 module doesn't run when Target=cc26x0-cc13x0 but works when Target=simplelink

ahmedmibrahim, 2019-10-20:
Hello, 

I have E-byte E70-915T14S module that has CC1310F128 chip.

I wired it correctly, and I managed to program it with multiple examples using CCS, either from TI Resource explorer or Energia or Contiki-ng

The problem is: 
- When using contiki-ng, it doesn't work except when I make TARGET=simplelink, and it never works with TARGET=cc26x0-13x0 even with the hello-world example
- I have to use TARGET=cc26x0-13x0 to be able to run the cc26xx-web-demo example.

Note: I have a CC1350 Launchpad which works fine with either Target (Same program)

How it doesn't work:
- TARGET=simplelink gives the correct LOG on teraterm as shown:      
          [INFO: Main ] Starting Contiki-NG-develop/v4.4-dirty
          [INFO: Main ] - Routing: RPL Classic
          [INFO: Main ] - Net: sicslowpan
          [INFO: Main ] - MAC: CSMA
          [INFO: Main ] - 802.15.4 PANID: 0xabcd
          [INFO: Main ] - 802.15.4 Default channel: 26
          [INFO: Main ] Node ID: 17175
          [INFO: Main ] Link-layer address: 0012.4b00.1884.4317
          [INFO: Main ] Tentative link-local IPv6 address: fe80::212:4b00:1884:4317
          [INFO: CC13xx/CC26xx] Operating frequency on Sub-1 GHz
          [INFO: CC13xx/CC26xx] RF: Channel 26
          [INFO: CC13xx/CC26xx] Node ID: 17175
          Hello, world

- Meanwhile, Target=cc26x0-13x0 gives a blank screen, I probed the UART pins on the chip and the UART TX (from the chip side) is held low. It seems like the platform didn't initialize successfully (platform_init_stage_one, platform_init_stage_two).

- I tried resetting multiple time with no hope.

Any Idea why?

ahmedmibrahim, 2019-10-20:
Just Bypassing this section got it working ... 

  /* Switch LF clock source to the LF XOSC if required */
  if(ti_lib_osc_clock_source_get(OSC_SRC_CLK_LF) == OSC_XOSC_LF) {
    ti_lib_osc_clock_source_set(OSC_SRC_CLK_LF, OSC_XOSC_LF);

    /* Wait for LF clock source to become XOSC_LF */
    while(ti_lib_osc_clock_source_get(OSC_SRC_CLK_LF) != OSC_XOSC_LF);

    /* Disable the LF clock qualifiers */
    ti_lib_ddi_16_bit_field_write(AUX_DDI0_OSC_BASE, DDI_0_OSC_O_CTL0,
                                  DDI_0_OSC_CTL0_BYPASS_XOSC_LF_CLK_QUAL_M |
                                  DDI_0_OSC_CTL0_BYPASS_RCOSC_LF_CLK_QUAL_M,
                                  DDI_0_OSC_CTL0_BYPASS_RCOSC_LF_CLK_QUAL_S,
                                  0x3);
  }

I appreciate if you can explain what happened

chenek, 2019-11-06:
It seems that there is something wrong with 32K external crystal of your CC1310 module. Mark this code  would by pass the crystal issue.

ahmedmibrahim, 2019-11-06:
mmm it's working with simplelink, doesn't simplelink use 32K crystal as well?

chenek, 2019-11-06:
Maybe it will bypass this part when you build with TARGET=SimpleLink

alexrayne, 2020-07-18:
Hallow!
this was spoted far time ago here [c26xx:prop-mode:rf on timeout - prevents code hung when OSC not start #2351](https://github.com/contiki-os/contiki/pull/2351)