contiki-ng/contiki-ng #1104: Contiki Simplelink Platform does not compile with Code Composer Studio

chris-ho, 2019-10-25:
None of the standard examples compiles succesful for the simplelink platform using the CCS. I followed the Wiki setup completely. 

Digging deeper into this issue brought up that the Makefiles using unsupported commands (not shipped with TI's Cygwin plugin of CCS). Namely tr, cut and grep.

In Makefile.cc13xx-26xx for instance we have two statements:

`DEVICE_LC := $(shell echo "$(DEVICE)" | tr A-Z a-z)`
`DEVICE_FAMILY_LC := $(shell echo "$(DEVICE_FAMILY)" | tr A-Z a-z)`

The consequence is that some variable are not set and hereafter the build fails (see attachment) because required include paths remain blank or incomplete.

[image]

chenek, 2019-11-06:
Try to use "TARGET = cc26x0-cc13x0" instead of "TARGET = simplelink" in your Makefile to see if it fixes your issue.

CONTIKI_PROJECT = hello-world

TARGET = cc26x0-cc13x0
BOARD = sensortag/cc2650
CFLAGS += -g

all: $(CONTIKI_PROJECT)

CONTIKI = ../..
include $(CONTIKI)/Makefile.include

chris-ho, 2019-11-06:
Hey @chenek thank you for this feedback that works - the only thing failing is the .elf to .hex conversion but that shoul not be an issue.

Then only the question which remains open is if there are plans to update the Makefiles with some conditionals for Windows/Linux to overcome the missing support of the listed commands in CCS.

chenek, 2019-11-06:
For elf to hex conversion, you will need to put srecord into your C:\Program Files\Git\bin

jeandudey, 2019-12-11:
I'm using CCS 9.2.0.00013, and this problem doesn't happen to me using the `simplelink` target with the `launchpad/cc1312r1` board. And as @chenek pointed out, SRecord needs to be in PATH - it can be modified on project properties -, or to don't mess up with that just put in  `C:\Program Files\Git\bin`.

@chris-ho it still can't compile?

chris-ho, 2019-12-13:
@jeandudey unfortunately I didn't had any success on this. Do you mind sharing your project which works fine for you?

even after adding SRECORD to my Path I get the same cygwin issue mentioned in my starting post:

LD        build/cc26x0-cc13x0/sensortag/cc2650/hello-world.elf
OBJCOPY   build/cc26x0-cc13x0/sensortag/cc2650/hello-world.elf --> build/cc26x0-cc13x0/sensortag/cc2650/hello-world.bin
OBJCOPY   build/cc26x0-cc13x0/sensortag/cc2650/hello-world.elf --> build/cc26x0-cc13x0/sensortag/cc2650/hello-world.i16hex
SREC_CAT  build/cc26x0-cc13x0/sensortag/cc2650/hello-world.i16hex --> build/cc26x0-cc13x0/sensortag/cc2650/hello-world.hex
../../arch/cpu/arm/Makefile.arm:73: recipe for target 'build/cc26x0-cc13x0/sensortag/cc2650/hello-world.hex' failed
/cygwin/sh: 1: grep: not found
/cygwin/sh: 1: cut: not found
/cygwin/sh: 1: grep: not found
/cygwin/sh: 1: cut: not found
process_begin: CreateProcess(NULL, srec_cat build/cc26x0-cc13x0/sensortag/cc2650/hello-world.i16hex -intel -o build/cc26x0-cc13x0/sensortag/cc2650/hello-world.hex -intel, ...) failed.
gmake: *** [build/cc26x0-cc13x0/sensortag/cc2650/hello-world.hex] Error 2
rm build/cc26x0-cc13x0/sensortag/cc2650/obj/fault-handlers.o build/cc26x0-cc13x0/sensortag/cc2650/obj/startup_gcc.o build/cc26x0-cc13x0/sensortag/cc2650/hello-world.i16hex hello-world.o
gmake: Target 'all' not remade because of errors.

**** Build Finished ****

chenek, 2019-12-19:
I suppose you need to install grep and cut on your Cygwin.

chris-ho, 2020-01-02:
@chenek, that's something I already tried.

After raising this issue as well in the TI E2E forum I got the confirmation that cut and grep has never been part of the cygwin installation which is shipped with Code Composer Studio.

So I tried to get them from a separate cygwin installation which unfortunately raised other issue which i can't remember in detail (more the less other commands are missing then :( )

If you have a proposal how I can find the mathing binaries of cut and grep that would be appreciated.

chenek, 2020-01-02:
I suppose you need to run Cygwin setup and make sure you install grep.
[image]

chris-ho, 2020-01-02:
@chenek 

Thank you finally installing cygwin from scratch (destination does not matter) and adding it to the PATH solved the issue.

closed 2020-01-02 by chris-ho

chenek, 2020-01-02:
Cool and it’s good to know the issue is solved!