contiki-ng/contiki-ng #1113: Sleep mode for zolertia (z1) cooja

philok93, 2019-10-31:
Hello,
i just converted my contiki v3 to contiki-ng and I am simulating z1 motes with csma driver in MAC layer. I see that radio is 100% ON during simulation. I know that some layers are merged like nullrdc+csma. However, I want to use a normal scenario where motes have radio off for most of the time and wake up when they receive a packet.In old contiki, I used contiki_mac driver to turn off radio for most of the time. Is that possible with contiki-ng? Or having radio on 100% is normal for a mote like z1? My goal is to measure the energy (and how much time they are ON) of nodes when they send too many packets.
Thank you!

atiselsts, 2019-11-05:
The CSMA MAC in Contiki-NG does not do duty cycling. Use TSCH if you want radio duty cycling. Or, if you want to keep CSMA, turn off the MAC layer from the application yourself with `NETSTACK_MAC.off()`.

closed 2019-11-05 by atiselsts

Dongddddong, 2020-08-02:
im a beginner. when i try to add `NETSTACK_MAC.off()` at the bottom of a 'while' loop and add `NETSTACK_MAC.on()` after `PROCESS_WAIT_UNTIL_EVENT()` in the loop, i find that there are no transmit  in a udp conmmunication. how do I get the communication back to normal?
thank you