contiki-ng/contiki-ng #1114: Can I use cfs to read/write sd card?

mttbx, 2019-11-03:
I have may files(about 50000). Is it a good idea to use cfs filesystem?
And Is there some protections in cfs when the power suddenly being blocked out?

mttbx, 2019-11-03:
The next two question are more important! I still need to use this filesystem to my eeprom.

nvt, 2019-11-04:
This is more of a support question rather than an issue for the project, so I am closing it.

Regarding the questions: I would not recommend it in your scenario because it is not designed to handle thousands of files efficiently. Instead, it was designed for a number of sensor networking use cases, such as sensor logs, debugging logs, packet queues, and routing tables. None of these use cases entailed that a large number of files would be needed. Additionally, to keep it simple and very small in RAM & ROM, it does not provide integrity guarantees in case of power loss or faulty sectors.

closed 2019-11-04 by nvt