contiki-ng/contiki-ng #1116: Maybe a bug in function find_next_record

mttbx, 2019-11-04:
Just read the code of cfs, I find that maybe there's a bug in cfs.
```
In function find_next_record: 
for(processed = 0; processed < log_records; processed += batch_size) {
      batch_size = log_records - processed >= preferred_batch_size ?
        preferred_batch_size : log_records - processed;

      COFFEE_READ(&indices, batch_size * sizeof(indices[0]),
                  absolute_offset(log_page, processed * sizeof(indices[0])));
      for(log_record = 0; log_record < batch_size; log_record++) {
        if(indices[log_record] == 0) {
          log_record += processed;
          break; <------------------------------------------ shoud we just use return log_record?
        }
      }
    }
```

mttbx, 2019-11-04:
Break can only jump out the inner loop

mttbx, 2019-11-04:
It may never find out because the following code:
```
preferred_batch_size = log_records > COFFEE_LOG_TABLE_LIMIT ?
    COFFEE_LOG_TABLE_LIMIT : log_records;  <------ about 4, much smaller then COFFEE_LOG_TABLE_LIMIT
```

g-oikonomou, 2020-11-25:
@nvt can you comment on this please and perhaps either label as a bug or close?

closed 2020-11-27 by mttbx