contiki-ng/contiki-ng #1118: contiki timer

apurva1992, 2019-11-04:
Hi, 

I am using contiki-6lbr on my CC1310 custom board. In my application code, I periodically(every 5 second) perform an action. Code is something like below:

PROCESS_THREAD(udp_client_process, ev, data)
{
    PROCESS_BEGIN();
    etimer_set(&et_periodic, 5 * CLOCK_SECOND);
    // Initialization code
   
    while(1)
    {

        PROCESS_YIELD();

        if(ev == PROCESS_EVENT_TIMER && data == &et_periodic)
        {
            //Perform some action
            //PRINTF statements
        }
    }
}

Code is running perfectly fine for initial few days. After few days of non-stop working, code behaves weirdly and doesn't enter if condition of et_periodic timer. The code is still working as few other threads are still printing the logs. It seems that Contiki timer service thread is hanging after few days. However, I am not able to figure out the root cause. Any leads would be highly appreciated.

Thank you!
Apurva

debug-ito, 2019-11-05:
It sounds like there is a bug in clock implementation of the platform. If you use arch/cpu/simplelink-cc13xx-cc26xx, the bug is probably in dev/clock-arch.c. Maybe the bug is related to wrapping of time (`clock_time_t` or internal counter in the platform), but I'm not so sure about it.

apurva1992, 2019-11-06:
Thank you for your response.
I checked clock_time_t which is uint32_t type. With each CLOCK_SECONDS consisting of 128 ticks, clock_time_t should not wrap for another 1 year. However, this issue I am seeing after one week of continuous run.

joakimeriksson, 2019-11-06:
Should you not also schedule the timer in the loop? I guess you did not show us that part of the code? Do you have a set / reset in the loop?

apurva1992, 2019-11-07:
Hi joakimeriksson,
Thanks for your response
Indeed, I am setting the timer in the loop as well.

`PROCESS_THREAD(udp_client_process, ev, data)
{
    PROCESS_BEGIN();
    etimer_set(&et_periodic, 5 * CLOCK_SECOND);
    // Initialization code

   while(1)
   {
        PROCESS_YIELD();

        if(ev == PROCESS_EVENT_TIMER && data == &et_periodic)
        {
            //destination_reachable is a function to check if node is connected to Border-Router
            destination_can_be_reached = destination_reachable();
            if (destination_can_be_reached)
            {
                //Perform some action
                //PRINTF statements
                etimer_set(&et_periodic, 5 * CLOCK_SECOND);
            }
            else
            {
                 //Perform some action
                 //PRINTF statements
                 etimer_set(&et_periodic, 1 * CLOCK_SECOND);
            }
        }
    }
}`

yatch, 2019-12-04:
One general comment; we need to narrow it down further.

>  It seems that Contiki timer service thread is hanging after few days.

If  `etimer` service thread (`etimer_process`) stops working entirely, other processes using `etimer` should be affected as well as your application program. Under `os/net`, you can find several `struct etimer` declarations, where `etimer` is used. Then, for instance, if you're using TSCH, you may want to check whether the 6LBR sends EBs or not (see `tsch_send_eb_process`).

yatch, 2019-12-04:
By the way,

```
    while(1)
    {
        PROCESS_YIELD();

        if(ev == PROCESS_EVENT_TIMER && data == &et_periodic)
        {
```

I would write it like this:

```
while(1) {
    PROCESS_WAIT_UNTIL(etimer_expired(&et_periodic));

...
```

It seems safer to check the timer expiration with `etimer_expired` than seeing the passed `data`.

alexrayne, 2020-05-13:
> It seems safer to check the timer expiration with etimer_expired than seeing the passed data.
check `ev == PROCESS_EVENT_TIMER` goes much faster vs call `etimer_expired`, so good to keep this check

closed 2020-11-25 by g-oikonomou