contiki-ng/contiki-ng #1119: DAG root switch parent? (maybe bug)

philok93, 2019-11-04:
Hi all,

I was trying to simulate a dense network with udp-clients and udp-server. I used the rpl-udp example with some modifications. While I was simulating in cooja, I noticed that root node 1 (udp-server) starts probing and selects a parent which causes loop. Why is this happening? Is it a bug that root tries to select a parent? I know that not all nodes have route in the network cause is very dense. 
[image]

I tried to look into the code of rpl-icmp6.c but didn't find anything that can explain this problem. Check the screenshot above. 

Thanks for your help.

philok93, 2019-11-04:
I found the problem should be in the following function:

```
void rpl_process_dio(uip_ipaddr_t *from, rpl_dio_t *dio)
{

  if(!curr_instance.used && !rpl_dag_root_is_root()) {
    /* Attempt to init our DAG from this DIO */
    if(!process_dio_init_dag(dio)) {
      LOG_WARN("failed to init DAG\n");
      return;
    }
  }
```

During simuation, there is a point where root receives a DIO and decides to return 0 from rpl_dag_root_is_root. How is this possible? Maybe its rank changes but can't understand why.
Appreciate any help.

debug-ito, 2019-11-07:
Hi,

This is weird indeed. So, before the node 1 receives a DIO and switches its parent, the node 1 should have set its rank to something other than `ROOT_RANK`. When and why did that happen? Note that `ROOT_RANK` equals to the `min_hoprankinc`, which is different for different Objective Functions.

philok93, 2019-11-07:
Hi,
I managed to solve the issue. The problem was that I tried to send a custom rpl packet to server but the server didn't parse it well (was trying to parse it several times) and this caused like overflow/unexpected behaviour of changing its rank other than 128. The solution was to fixed the server to parse correctly the packet (changed the correct size of fields in the packet) and now this behaviour doesn't happen. So, I close this thread. Thanks for your help.

closed 2019-11-07 by philok93