contiki-ng/contiki-ng #1120: very-sleepy-demo watchdog trigger after calling NETSTACK_MAC.off()

ahmedmibrahim, 2019-11-06:
Hello, 
On cc1350 LP, using very_sleepy_demo, whenever NETSTACK_MAC.off() is called, the watchdog triggers and the device restarts.
By adding a printf statement after NETSTACK_MAC.off(), I concluded the function is executed fully before the watchdog triggers.

Any idea why this happens and what is the solution?

arurke, 2019-11-06:
* Have you prohibited shutdown of some power domains? If so it might be caused by https://github.com/contiki-ng/contiki-ng/issues/1075.
* Seems unlikely, but could it be a regression from https://github.com/contiki-ng/contiki-ng/pull/1074? Maybe try to checkout a commit before this was merged to verify.

g-oikonomou, 2019-11-06:
This example makes a number of assumptions about other things going on elsewhere in the code. One of the things it assumes is the behaviour changed with #1052.

It is very likely that you are looking at a bug in this example's code, which has been untouched for over two years IIRC and I am sure would benefit from a bit of love.

I'll flag this one as a bug to make sure it doesn't fall through any cracks.

ahmedmibrahim, 2019-11-07:
@arurke is right, I re-cloned and it worked
sorry for false alarm, I think there was one power module on

closed 2019-11-07 by ahmedmibrahim