contiki-ng/contiki-ng #1121: Upload issue for openmote-b board

glegall, 2019-11-06:
Hi,

I am using Contiki NG with the OpenMote B board, and when I saw there was an effort to officially support it, I switched to the official platform definition for it.

Then I encountered an issue, the `cc2538-bsl.py` script would fail with the following error:

```
ERROR: Timeout waiting for ACK/NACK after 'Synch (0x55 0x55)'
```

In the definition of the board, especially the file board.h <[https://github.com/contiki-ng/contiki-ng/blob/develop/arch/platform/openmote/openmote-b/board.h#L64](url)>, the "enable bootloader" pin is changed from A6 to A7. If lines 64 to 70 were removed, the pin would be defined here <[https://github.com/contiki-ng/contiki-ng/blob/develop/arch/platform/openmote/contiki-conf.h#L80](url)>.

If pin A7 is selected, I have to physically wire it to ground with a jumper wire during the flashing process for it to succeed. Luckily enough, pin A7 is exposed on the board's header.

Also, I needed to pass the `--bootloader-invert-lines` option to the `cc2538-bsl.py` script for the flashing process to work.

If pin A6 is selected in `board.h` for entering the bootloader, together with the`--bootloader-invert-lines` argument, from what I understand the `cc2538-bsl.py` script can control the "enter bootloader mode" with the RTS signal of the serial line. So there is no need for extra wiring, in that case.

So, basically, the following changes fixed the issue I was having:

* Remove line 64 to 70 in file `arch/platform/openmote/openmote-b/board.h`.
* Change line 32 of the `openmote` Makefile (see <[https://github.com/contiki-ng/contiki-ng/blob/develop/arch/platform/openmote/Makefile.openmote#L32](url)>) into:

```
BSL_FLAGS += -e
ifeq ($(BOARD),openmote-b)
        BSL_FLAGS += --bootloader-invert-lines
endif
BSL_FLAGS += -w -v -b 450000
```

What do you think about this? Did anyone had a similar issue with OpenMote B?

Thank you.

g-oikonomou, 2019-11-06:
@wallb any thoughts?

wallb, 2019-11-12:
I have tried with the changes you suggested, @glegall, on two different motes but cannot get it to work. Maybe the explanation for our different results is along the lines of what is mentioned here:
https://github.com/RIOT-OS/RIOT/issues/11684

Having to do the manual wiring is unfortunate, but perhaps it is wise to have that as the default way to have it work on all motes and possibly add using A6 as a build option.

glegall, 2019-11-21:
﻿Thank you very much for your replies.

Indeed, the issue you are referring to in RIOT seems closely related.

In there they say that the manufacturer made some changes in the MSP430 that is used to control the flashing process.

My guess would be that we have different versions of the board, or at least of MSP firmware.

Just being curious, what is the revision of your board? When I look at the bottom of mine, near the USB plug it reads "`HW Rev. 1.4`".

If this assumption is correct, I guess the definition of the board in Contiki should then reflect what the boards that are sold from now on require. Maybe your version?

wallb, 2019-11-25:
It says `HW Rev. 1.4` on the motes we have. So I guess the difference is the MSP firmware.

closed 2020-04-15 by g-oikonomou