contiki-ng/contiki-ng #1126: how to enter low power mode in cc2538dk?

xiangxiangjiang, 2019-11-13:
I use the hello world programe  to test the low power mode.But the current is about 27mA. How should I do to enter the low power mode?
Thank you
My setting is 

#ifndef LPM_CONF_ENABLE
#define LPM_CONF_ENABLE       1 /**< Set to 0 to disable LPM entirely */
#endif

/**
 * \brief Maximum PM
 *
 * The SoC will never drop to a Power Mode deeper than the one specified here.
 * 0 for PM0, 1 for PM1 and 2 for PM2
 */
#ifndef LPM_CONF_MAX_PM
#define LPM_CONF_MAX_PM       2//1
#endif

xiangxiangjiang, 2019-11-14:
Is the default config low power mode? cc2538dk use the default config to run hello world program,the current is about 27mA. Did anyone use cc2538dk to sucess getting into low power mode? Is the current down to less 1mA? I'm very confused about that. QAQ

matiya, 2019-11-29:
I also had issues with this. According [this line](https://github.com/contiki-ng/contiki-ng/blob/f87846cfa082e4b4cc43f8827ec9eaebcbbaa3ba/arch/cpu/cc2538/lpm.c#L236), there are series of peripherals that can block the PM1+.
 
The peripherals that can block are:
 1) UART
 2) USB
 3) PWM
 4) PKA
 5) Crypto

The define:

`#define CC2538_CONF_QUIET           1`

takes care of the first two I believe

xiangxiangjiang, 2019-11-29:
> 我对此也有疑问。根据[这条线](https://github.com/contiki-ng/contiki-ng/blob/f87846cfa082e4b4cc43f8827ec9eaebcbbaa3ba/arch/cpu/cc2538/lpm.c#L236)，有一系列外围设备可能会阻塞PM1 +。
> 
> 可能阻塞的外围设备有：
> 
> 1. 串口
> 2. USB
> 3. 脉宽调制
> 4. PKA
> 5. 加密货币
> 
> 定义：
> 
> `#define CC2538_CONF_QUIET 1`
> 
> 我相信前两个

did you solve it? I tryed many methods,but they didn't work. Contiki 3.0 works fine,but its net works less better than contiki-ng. So, I choose contiki-ng,but its low power mode seems not to work.

matiya, 2019-11-30:
I could get the nodes into low power mode with the define that I wrote. Since then we change the MAC protocol to TSCH and I don't think it works with LPM, but I might be wrong.

I would advise you to use gdb+renode and set a breakpoint in the lpm_enter() function. Then you can explore what condition is blocking LPM.

xiangxiangjiang, 2019-12-02:
@matiya 
Thank you. I define:
`#define UART_CONF_ENABLE            0`
`#define CC2538_CONF_QUIET           1`
I used the hello word program to test LMP, but it didn't work. 

hello world.c is 
`#include "contiki.h"
#include "sys/energest.h"
#include "dev/leds.h"
#include <stdio.h> /* For printf() */
/*---------------------------------------------------------------------------*/
PROCESS(hello_world_process, "Hello world process");
AUTOSTART_PROCESSES(&hello_world_process);

PROCESS_THREAD(hello_world_process, ev, data)
{
  static struct etimer timer;
  static int i=5;
  PROCESS_BEGIN();
  leds_off(LEDS_ALL);
  /* Setup a periodic timer that expires after 10 seconds. */
  etimer_set(&timer, CLOCK_SECOND * 1);
  while(i) {
    /* Wait for the periodic timer to expire and then restart the timer. */
    PROCESS_WAIT_EVENT_UNTIL(etimer_expired(&timer));
    etimer_reset(&timer);
  }
  PROCESS_END();
}`

the cc2538-conf.h is
`
#ifndef CC2538_CONF_H_
#define CC2538_CONF_H_
/*---------------------------------------------------------------------------*/
/**
 * \name CFS configuration
 *
 * @{
 */
#ifndef COFFEE_CONF_SIZE
#define COFFEE_CONF_SIZE            (4 * COFFEE_SECTOR_SIZE)
#endif
/** @} */
/*---------------------------------------------------------------------------*/
/**
 * \name CC2538 System Control configuration
 *
 * @{
 */
#ifndef SYS_CTRL_CONF_OSC32K_USE_XTAL
#define SYS_CTRL_CONF_OSC32K_USE_XTAL   0 /**< Use the on-board 32.768-kHz crystal */
#endif
/** @} */
/*---------------------------------------------------------------------------*/
/**
 * \name Watchdog Timer configuration
 *
 * @{
 */
#ifndef WATCHDOG_CONF_ENABLE
#define WATCHDOG_CONF_ENABLE        1 /**< Enable the watchdog timer */
#endif
/** @} */
/*---------------------------------------------------------------------------*/
/**
 * \name USB 'core' configuration
 *
 * Those values are not meant to be modified by the user, except where stated
[... 225 lines cut ...]
#endif
/** @} */
/*---------------------------------------------------------------------------*/
/**
 * \name Security
 *
 * @{
 */
#ifndef CRYPTO_CONF_INIT
#define CRYPTO_CONF_INIT        1 /**< Whether to init cryptoprocessor */
#endif

#ifndef AES_128_CONF
#define AES_128_CONF            cc2538_aes_128_driver /**< AES-128 driver */
#endif

#ifndef CCM_STAR_CONF
#define CCM_STAR_CONF           cc2538_ccm_star_driver /**< AES-CCM* driver */
#endif
/** @} */
/*---------------------------------------------------------------------------*/
#endif /* CC2538_CONF_H_ */
/*---------------------------------------------------------------------------*/
/** @} */
`

closed 2020-04-15 by g-oikonomou

ameseguerITI, 2023-08-26:
Dear @xiangxiangjiang and @g-oikonomou I am suffering the same problem. Using contiki-ng and cc-2538 base board, the consumption is always 26mA. CONF_QUIET and LPM are enabled. May you give me any advice?