contiki-ng/contiki-ng #1128: Cannot build hello-world for CC26x2R1.

chenek, 2019-11-14:
I git to develop branch and try to build hello-world for CC26x2R1 with "make TARGET=simplelink BOARD=launchpad/cc26x2r1". However, I get the following build error.

`$ make TARGET=simplelink BOARD=launchpad/cc26x2r1
  MKDIR     build/simplelink/launchpad/cc26x2r1/obj
  CC        /cygdrive/c/contiki-ng/arch/cpu/simplelink-cc13xx-cc26xx/./ccfg-conf.c
arm-none-eabi-gcc.exe: error: /cygdrive/c/contiki-ng/arch/cpu/simplelink-cc13xx-cc26xx/./ccfg-conf.c: No such file or directory
arm-none-eabi-gcc.exe: fatal error: no input files
compilation terminated.
make: *** [/cygdrive/c/contiki-ng/arch/cpu/simplelink-cc13xx-cc26xx/Makefile.cc13xx-cc26xx:138: build/simplelink/launchpad/cc26x2r1/obj/ccfg-conf.o] Error 1
`

What am I doing wrong and how to fix this?

ahmedmibrahim, 2019-11-15:
Did you correctly clone to contiki-ng latest repository and ran:
sudo git submodule update --init --recursive?

chenek, 2019-11-15:
Yes, I did it and still see the problem.

andak, 2019-12-02:
You're probably observing [this bug](https://bugs.launchpad.net/gcc-arm-embedded/+bug/1282943); the arm-none-eabi toolchain does not understand cygwin-paths. As an alternative, [MinGW](http://mingw.org) is supposed to work according to the comments in the link.

chenek, 2019-12-03:
@andak Thanks for your reply. I agree with you this might be toolchain issue since I can use Ubuntu to set up build environment to build the example successfully with the same make command.

closed 2019-12-03 by chenek