contiki-ng/contiki-ng #1129: BLE with simplelink-platform

0608rohit, 2019-11-14:
Hello 

I am using demo code of contiki-ng's  "contiki-ng_4.3/examples/platform-specific/cc26x0-cc13x0/cc26x0-demo.c" and make changes according to simplelink platform. 

Also, I enabled BLE advertisement on CC1352r1-lunchxl successfully.

Now, I want establish connection between CC1352r1-lunchxl and Bluetooth enabled device  through "LightBlue" android application and Also, want to send specific data over Bluetooth.

Thanks in advance.

chenek, 2019-11-15:
You might need to try to use TI DMM project to run TI BLE Stack and Contiki-NG concurrently to support this.

0608rohit, 2019-11-25:
Hello YK chenek,
I want to start BLE stack (DMM) and contiki-ng Concurrently
Documentation, Link or source code would be helpful.

closed 2019-11-25 by 0608rohit

chenek, 2019-11-25:
Try to refer to http://dev.ti.com/tirex/explore/node?node=AMu-cvNgpJsLYI.IMLUI.A__pTTHBmu__LATEST

chenek, 2019-11-25:
I am afraid TI DMM can only support TI RTOS project. I am sorry for misleading in my previous reply.