contiki-ng/contiki-ng #1131: cc13x0 deep sleep mode power consumption is still high

ahmedmibrahim, 2019-11-15:
Hello, 
I ran the very-sleepy-demo on cc1350 LAUNCHXL, when the device is in deep sleep mode, the current consumption is ~2.5mA which is still pretty high.
- Are there any other modules that are still powered on in the deep-sleep mode?
- How to power them down?

[image]

rajeev1986, 2019-11-15:
@ahmedmibrahim, that's because the TSCH scanning thread and the scheduler are still running keeping the MCU active, preventing it from deep sleep. My guess is if you compile the **very-sleepy-demo**  with `MAKE_MAC = MAKE_MAC_NULLMAC` and `MAKE_NET = MAKE_NET_NULLNET` the device power consumption will be in tens of microamps rather than in milli.  However, if you want to keep the default MAC and network running, you will have to explicitly turn off the TSCH before going to deep sleep and turning it back on after waking up.

ahmedmibrahim, 2019-11-16:
Thanks for your response, 
- I'm not using TSCH at all.
- I tried building with NULLMAC and NULLNET but it failed, do you know how to turn TSCH?

ahmedmibrahim, 2019-11-23:
@g-oikonomou TI's standby example shows < 10uA of standby current, do you know why the verysleepy demo deepsleep period is consuming ~140uA?

ahmedmibrahim, 2019-12-05:
@chenek  Do you have any idea how to reduce power here?

chenek, 2019-12-05:
Try to check you call ext_flash_close to shutdown external flash.

ahmedmibrahim, 2019-12-07:
Tried turning off ext flash, no change. I think this example doesn't use external flash.
Are there any modules powered on on the MCU itself? I'm just running the demo without any changes.

chenek, 2019-12-07:
Can you attach a picture of your LAUNCHXL-CC1350 when you measure power consumption?

ahmedmibrahim, 2019-12-07:
I have verified the setup with TI on this thread
https://e2e.ti.com/support/wireless-connectivity/sub-1-ghz/f/156/t/849367
[image]

I get 5.4µA consumption when I ran TI demo (pinStandby) --> I'm confident of my setup

chenek, 2019-12-07:
Try to keep GND/TX/RX jumpers mounted and test again.

ahmedmibrahim, 2019-12-07:
NO difference (this will make a difference at lower currents), as I told you, I get 5.4uA using the same setup by just running another example (TI example).

There must be a power module that is still on, this module is not the ext flash (It must be on the MCU itself).

chenek, 2019-12-07:
Since you use XDS110 on MSP432 launchpad, I think you should connect GND/TX/RX on CC1350 side of LAUNCHXL-CC1350 to XDS110 side of MSP432 launchpad when you do power consumption measurement.

ahmedmibrahim, 2019-12-08:
Hi @chenek , 
- Do you know what processes are still running in Deep Sleep Mode in this example? 
- Do you know their power consumption?
- Do you know how to turn them off?

chenek, 2019-12-08:
As I know, there’s o process running during deep sleeping mode. Do you try to remove all jumpers between XDS110 and CC1350, input voltage from 3V3 and GND pins in the bottom of LAUNCHXL-CC1350 and don’t connect JTag to do power measurement?

ahmedmibrahim, 2019-12-08:
Yes with nothing connected at all (not even the RESET pin) I get ~ 85uA during Deep Sleep Mode
[image]

Measurement:
[image]

Also sometimes it drops to 2mA only ... any idea why?
[image]

chenek, 2019-12-09:
It seems that you use energy trace on MSP432 launchpad to do power measurement. If so, it means you still have JTag connected. I would suggest you to use power analyzer or scope to do power measurement.

ahmedmibrahim, 2019-12-09:
Ok, do you know why the program drops to a higher current value (~2mA)?

chenek, 2019-12-09:
I suspect CC1350 wakes up in this case but I am not sure what happens in your case.

ahmedmibrahim, 2019-12-09:
I think lpm_drop doesn't proceed to deep_sleep_mode, but just the sleep_mode, because a module advertised max_pm supported = LPM_MODE_SLEEP 

I wonder if there is any means to ensure all modules are capable of LPM_MODE_DEEP_SLEEP before calling process_idle()

ahmedmibrahim, 2019-12-09:
Found it --> I2C Pins are moving randomly up and down (I don't know why), when they are low, more leakage through the pull up Resistors (3.3K).
Leakage = 3.3V / 3.3KOhm x 2 (SDA/SCL) = 2 mA  :)

closed 2019-12-09 by ahmedmibrahim

chenek, 2019-12-09:
You should set unused pin to PIN_UASSIGNED in your board file.

ahmedmibrahim, 2019-12-11:
hey @chenek , 
PIN_UNASSIGNED is for simplelink Target,
very_sleepy_demo uses cc26xx-13xx, Can you explain how to do that on this target?
I tried adding pins to BOARD_UNUSED_PINS list in board.h file and it didn't help
I still have some IOs toggling high and low when moving from deep_sleep to active

Also please explain (in detail) how to set the default value of a PIN in Deep_sleep_mode. Example: I want DIO30 to be pulled up in sleep, while DIO25 should be pulled low.

chenek, 2019-12-12:
I suppose you should set unused pins to PIN_UNASSIGNED in LAUNCHXL-CC1350.h.

NeoN7777, 2019-12-13:
Hi @chenek 
Do you know what is appropriate power consumption for CC1310 when using TSCH? I have 150-200 uA on a router node, is there any way to reduce the power? Thanks.

chenek, 2019-12-13:
I think 150-200 uA on a router node is pretty low.

NeoN7777, 2019-12-13:
> I think 150-200 uA on a router node is pretty low.

So, is there no way to make it 20-40uA with router functions, maybe with some limited abilities?

chenek, 2019-12-13:
I don’t think so.

NeoN7777, 2019-12-13:
> I don’t think so.

Okay :(
Thank you

ahmedmibrahim, 2019-12-13:
A router node cannot go in Deep Sleep mode while routing at the same time
That stand by mode usually consumes ~6-7mA (radio is on, not transmitting), during transmission you will notice current spikes (sudden high current for a very short time) that may reach 22mA (check the datasheet for accurate numbers)

ahmedmibrahim, 2019-12-13:
BTW, for deep sleep of cc1350 launchpad, 
1- make sure you do JTAG Reset (full reset) after flashing a new image (don't rely on CCS reset) as it gives 100uA leakage
2- my current current consumption is ~8uA in Deep sleep mode (measured with energyTrace) which is still not as the datsheet value (0.7) for cc1350 and 7nA for external Flash

Please let me know if you managed to optimize it further, I desperately need it

chenek, 2019-12-13:
I think it’s normal to see ~8uA in Deep sleep mode on Launchpad since there are other components on Launchpad. datsheet value (0.7) for cc1350 is CC1350 only and not including other components.

ahmedmibrahim, 2019-12-14:
I can't see anything else more than the MCU and the Ext Flash... the RF radio should be off and its circuitry.
Anyways, can you answer my question about:

> PIN_UNASSIGNED is for simplelink Target,
very_sleepy_demo uses cc26xx-13xx, Can you explain how to do that on this target?
I tried adding pins to BOARD_UNUSED_PINS list in board.h file and it didn't help
I still have some IOs toggling high and low when moving from deep_sleep to active

> Also please explain (in detail) how to set the default value of a PIN in Deep_sleep_mode. Example: I want DIO30 to be pulled up in sleep, while DIO25 should be pulled low.

chenek, 2019-12-14:
According to your current power consumption, I think you already set pins correctly.