contiki-ng/contiki-ng #1134: Include dir order problem in native

pipex, 2019-11-22:
Hello,

I have a problem when building a project for native platform. I include a copy of my Makefile for reference below

The file [cpu/Makefile.native](https://github.com/contiki-ng/contiki-ng/blob/9b2dfb0d1cbcb3500813508e92f8b662d73bfe9b/arch/cpu/native/Makefile.native#L20) adds `/usr/local/include` to the include paths. Given the order of operations in Makefile.include, the `-I/usr/local/include` flag ends up being the first one in the CC command call. 

This means that if I have in my project an include matching an include in /usr/local, the the build gives priority to the file in /usr/local

e.g. I have an event.h in my project, but the build never finds it because I have libevent installed as well

I get that I can fix it by adding `-I<src>` to CFLAGS before including the Makefile.include, but you would think it should suffice by setting PROJECTDIRS in the makefile.

```
CONTIKI_PROJECT = echo
all: $(CONTIKI_PROJECT)

TWO = ../../src
CONTIKI ?= ../../../contiki-ng

PROJECTDIRS += $(TWO)
PROJECT_SOURCEFILES += cbuf.c event.c

DEFINES += UIP_CONF_TCP=1
DEFINES += UIP_CONF_UDP=0

include $(CONTIKI)/Makefile.include
```

g-oikonomou, 2019-11-22:
I cannot immediately understand why we are explicitly doing `-I/usr/local/include`. I would suggest simply removing it and seeing how travis feels.

Having said all that, do you _really_ need to call you header `event.h`. I'd avoid using filenames that are also in-use by the toolchain. I would for example definitely avoid creating a header called `stdio.h` if you see what I mean.

pipex, 2019-11-22:
Hi George,

Thanks for the quick reply. Would you like me to submit a mini-pr with the updated Makefile?

Regarding your suggestion, I will consider renaming the file, however I didn't see much problem since there are no other files in the standard library called like that. I happenned to have installed libevent locally which caused the problem.

Thanks again

g-oikonomou, 2019-11-22:
Ah right, I am with you now. By all means go right ahead and create a PR and let's see what happens

g-oikonomou, 2019-11-22:
Fixed in #1135

closed 2019-11-22 by g-oikonomou