contiki-ng/contiki-ng #1136: The mqtt-client example causes cooja to die a terrible death

g-oikonomou, 2019-11-23:
To reproduce, create a new sim inside docker. Create two cooja motes: 1x border router and 1x mqtt-client. Follow the necessary steps in order for the BR to obtain a prefix and start the network.

If the MQTT client and the BR are outside each other's radio range, or while the BR is still searching for a prefix things work fine. As soon as they get within range though, cooja dies with this output:

```
[...]
     [java]  INFO [Thread-6] (Simulation.java:253) - Simulation main loop started, system time: 1574522544845
     [java]  INFO [Thread-7] (SerialSocketServer.java:478) - Forwarder: socket -> serial port
     [java]  INFO [Thread-5] (SerialSocketServer.java:425) - Client connected: /127.0.0.1
     [java] #
     [java] # A fatal error has been detected by the Java Runtime Environment:
     [java] #
     [java] #  SIGSEGV (0xb) at pc=0xc12f51ff, pid=7796, tid=0xc2073b40
     [java] #
     [java] # JRE version: OpenJDK Runtime Environment (8.0_222-b10) (build 1.8.0_222-8u222-b10-1ubuntu1~16.04.1-b10)
     [java] # Java VM: OpenJDK Server VM (25.222-b10 mixed mode linux-x86 )
     [java] # Problematic frame:
     [java] # C  [mtype679.cooja+0x1b1ff]  uip_icmp6_send+0x5d
     [java] #
     [java] # Failed to write core dump. Core dumps have been disabled. To enable core dumping, try "ulimit -c unlimited" before starting Java again
     [java] #
     [java] # An error report file with more information is saved as:
     [java] # /home/user/contiki-ng/tools/cooja/build/hs_err_pid7796.log
     [java] Java Result: 139
```

So it appears as though the problem will only occur while the mqtt client is trying to join the RPL network, or after it has done so.

The problem only appears with RPL lite, not with classic. The problem only appears if there is an mqtt-client in the simulation, other examples work fine.

Affects 4.2, 4.3 and latest `develop`

yatch, 2019-11-23:
I reproduced the issue. `SIGSEGV` occurred here, L239, where `dest` is `NULL`:
```
 229 void
 230 uip_icmp6_send(const uip_ipaddr_t *dest, int type, int code, int payload_len)
 231 {
 232   UIP_IP_BUF->vtc = 0x60;                                                                                                               233   UIP_IP_BUF->tcflow = 0;
 234   UIP_IP_BUF->flow = 0;
 235   UIP_IP_BUF->proto = UIP_PROTO_ICMP6;
 236   UIP_IP_BUF->ttl = uip_ds6_if.cur_hop_limit;
 237   uipbuf_set_len_field(UIP_IP_BUF, UIP_ICMPH_LEN + payload_len);
 238
 239   memcpy(&UIP_IP_BUF->destipaddr, dest, sizeof(*dest));
```
`uip_icmp6()` is called by `ping_parent()` in `mqtt-client.c`. I will do a little more research to see why `rpl-classic` doesn't cause this issue.

g-oikonomou, 2019-11-23:
It appears as though we have a global v6 address but we do not have a default route yet.

g-oikonomou, 2019-11-23:
Does this look like a valid fix @yatch ?

```
$ git diff -- examples/mqtt-client/
diff --git a/examples/mqtt-client/mqtt-client.c b/examples/mqtt-client/mqtt-client.c
index d725e46b9..51de53117 100644
--- a/examples/mqtt-client/mqtt-client.c
+++ b/examples/mqtt-client/mqtt-client.c
@@ -591,7 +591,8 @@ connect_to_broker(void)
 static void
 ping_parent(void)
 {
-  if(uip_ds6_get_global(ADDR_PREFERRED) == NULL) {
+  if(uip_ds6_get_global(ADDR_PREFERRED) == NULL ||
+     uip_ds6_defrt_choose() == NULL) {
     return;
   }
 
@@ -631,7 +632,8 @@ state_machine(void)
     LOG_DBG("Init\n");
     /* Continue */
   case STATE_REGISTERED:
-    if(uip_ds6_get_global(ADDR_PREFERRED) != NULL) {
+    if(uip_ds6_get_global(ADDR_PREFERRED) != NULL &&
+       uip_ds6_defrt_choose() != NULL) {
       /* Registered and with a public IP. Connect */
       LOG_DBG("Registered. Connect attempt %u\n", connect_attempt);
       ping_parent();
```

yatch, 2019-11-23:
@g-oikonomou Yes, `rpl-lite` defers selecting the preferred parent. I identified the code. I will open a PR.

yatch, 2019-11-23:
OK; you've already fixed the issue. Great. Can you put a NULL-pointer check to `uip_icmp6_send()`?

g-oikonomou, 2019-11-23:
OK great

here is what I have so far. Feel free to take stuff

```
$ git diff -- examples/mqtt-client/
diff --git a/examples/mqtt-client/mqtt-client.c b/examples/mqtt-client/mqtt-client.c
index d725e46b9..0f0451671 100644
--- a/examples/mqtt-client/mqtt-client.c
+++ b/examples/mqtt-client/mqtt-client.c
@@ -250,6 +250,17 @@ static const uint8_t mqtt_client_extension_count = 0;
 /*---------------------------------------------------------------------------*/
 PROCESS(mqtt_client_process, "MQTT Client");
 /*---------------------------------------------------------------------------*/
+static bool
+have_connectivity(void)
+{
+  if(uip_ds6_get_global(ADDR_PREFERRED) == NULL ||
+     uip_ds6_defrt_choose() == NULL) {
+    return false;
+  }
+
+  return true;
+}
+/*---------------------------------------------------------------------------*/
 static int
 ipaddr_sprintf(char *buf, uint8_t buf_len, const uip_ipaddr_t *addr)
 {
@@ -591,7 +602,7 @@ connect_to_broker(void)
 static void
 ping_parent(void)
 {
-  if(uip_ds6_get_global(ADDR_PREFERRED) == NULL) {
+  if(have_connectivity() == false) {
     return;
   }
 
@@ -631,7 +642,7 @@ state_machine(void)
     LOG_DBG("Init\n");
     /* Continue */
   case STATE_REGISTERED:
-    if(uip_ds6_get_global(ADDR_PREFERRED) != NULL) {
+    if(have_connectivity()) {
       /* Registered and with a public IP. Connect */
       LOG_DBG("Registered. Connect attempt %u\n", connect_attempt);
       ping_parent();

```

g-oikonomou, 2019-11-23:
> OK; you've already fixed the issue. Great. Can you put a NULL-pointer check to uip_icmp6_send()?

You go right ahead 😄

yatch, 2019-11-23:
You can reproduce the issue the following steps:

* downloading `issue-1136-rpl-lite.txt` attached here
* change their suffix to `.csc`
* load `issue-1136-rpl-lite.csc` from `cooja` and start the simulation
* run `make TARGET=cooja connect-router-cooja` from `examples/rpl-border-router`

You can see how `rpl-classic` behave in the same scenario with `issue-1136-rpl-classic.txt`.

[issue-1136-rpl-classic.txt](https://github.com/contiki-ng/contiki-ng/files/3882764/issue-1136-rpl-classic.txt)
[issue-1136-rpl-lite.txt](https://github.com/contiki-ng/contiki-ng/files/3882765/issue-1136-rpl-lite.txt)

yatch, 2019-11-23:
@g-oikonomou I got it; I'm going to open one.

g-oikonomou, 2019-11-23:
Fixed in #1137

closed 2019-11-23 by g-oikonomou