contiki-ng/contiki-ng #1145: TSCH base drift ppm for CC1310

hseokman, 2019-11-27:
arch/cpu/cc26x0-cc13x0/cc13xx-cc26xx-def.h defines TSCH_CONF_BASE_DRIFT_PPM as -977.
It is calculated for 10ms time slot.
Since cc1310 uses 40ms time slot, should this be changed accordingly?
Slot length 40000 usec which is 1311 ticks
1311 ticks is 40008.544922 usec
-8.544921 / 40008.544921 = -0.0002135774.
Therefore I think it needs to be -214 for cc1310.

I don't quite follow the calculation for the 10 ms timeslot. Should this be
-9.765625 / 10009.765625 = -0.00097560, so -976 instead of -977?

Also this base drift is not compensated for the first sleep.
tsch_slot_operation_start() from os/net/mac/tsch/tsch-slot-operation.c
does not call tsch_timesync_adaptive_compensate. There is no drift calculated for the association, but base drift rate still needs to accounted.

atiselsts, 2019-11-28:
I think you are right on all accounts. Can you make a PR?

hseokman, 2019-12-04:
I just created PR. I think I'm mistaken in PPM calculation. It's supposed to be delta/target, so -977 for the 2.4 Ghz is correct.

atiselsts, 2019-12-05:
Let's take the 10 millisecond slots.

My reasoning is this:
10009.765625 microseconds is the time duration of one uncompensated TSCH slot,

Now assume compensation with base drift 977 ppm. The actual slot length becomes:
 10009.765625 * (1 - 0.000977) = 9999.986083984375 microseconds - not bad.

Now assume compensation with base drift 976 ppm:
10009.765625 * (1 - 0.000976) = 9999.99609375 - better!

hseokman, 2019-12-05:
I guess the actual calculation depends on how the compensation math is done. At first I thought your way because that seems just natural about compensation, but the ppm is really 'Variance/Target'. 
I looked at the how the compensation work and it basically tries to add 'time_delta_usec * ppm' to the calculation every time its called. It also records remainder and tick conversion error every time to calculate the next compensation.
When the compensate method is called for the 10 ms, the device actually delays for 328 ticks which is 10009.765625 usec 
for drift -977 ppm
It adds 10000 * (-0.000977) = -9.77 usec for compensation
Error: 10009.765625 - 9.77 = 9999.995625
for drift -976 ppm
It adds 10000 * (-0.000976) = -9.76 usec for compensation
Error: 10009.765625 - 9.76 = 10000.005625

Therefore, -977 seems correct.

atiselsts, 2019-12-06:
Fixed in #1148, thanks!

closed 2019-12-06 by atiselsts