contiki-ng/contiki-ng #1146: TSCH Orchestra + UDP messages - Cooja Simulation

martincacereal, 2019-12-01:
Dear community;

I am working in posgraduate micro project about Wireless Sensor Networks, the objective of this porject is to masuere TSCH Orchestra performance with different topologies and number of nodes.

I decided to work with the example "timesync-demo"  because it had implemented the UDP massages and the TSCH network. Also I noticed the z1 plataform is excluded from the example Somebody could explian me why that?

Anyway, I tried with z1 platfom and for RPL LITE and TSCH mininal and it is working fine, but when I try to change to orchestra does not work. I quite new with contiki and contiki-ng so I have not much experience on them; so, I will explain what I am doing setp by setp.

First of all, I simulated on cooja "timesync-demo" but including z1 (I deleted it from the excluded platforms from the makefile). Also I had to do the following, I commented the #if line where the coordinator is set:

//#if CONTIKI_TARGET_COOJA 
  is_coordinator = (node_id == 1);
//#endif

After this the example was working well.

Also in the project-conf.h I added #define TSCH_CONF_EB_PERIOD     (4 * CLOCK_SECOND), because I found there is a bug in cooja (/* A bug in cooja causes many EBs to be missed at scan. Increase EB frequency to shorten the join process */)

That last one makes the join process faster.

After that, I made a "make clean" cleaning the TARGET used, in this case z1. After that, I excuted "make TARGET=z1 MAKE_WITH_ORCHESTRA=1".

Once I did the compilation, I went to cooja and I made a new simulation with  5 nodes, firt of all with a linear topology, and did not work. The only node could "join" to TSCH was the second one (the only one in the range of the coordinator). And actually I reckon it cannot join to the DODAG. If I put the nodes in the range of the coordinator every node can "join" to the TSCH but they cannot send messages with the root and coordinator (node ID:1).

Also in the LOG I can find some errors but I am trying to track them in contiki implementation with I did not succeed to find the problem (obviously, because I do not have too much expertise in this). Bellow I share erros logged and in the order which they were happenig:

00:01.152	ID:2	[ERR : TSCH Sched] ! add_link memb_alloc failed
00:01.264	ID:4	[ERR : TSCH Sched] ! add_link memb_alloc failed
00:01.329	ID:1	[ERR : TSCH Sched] ! add_link memb_alloc failed
00:01.662	ID:5	[ERR : TSCH Sched] ! add_link memb_alloc failed
00:01.815	ID:3	[ERR : TSCH Sched] ! add_link memb_alloc failed
00:07.320	ID:2	[ERR : TSCH Sched] ! add_link memb_alloc failed
00:50.794	ID:1	[ERR : TSCH Queue] ! add packet failed: 0 0x1c82 -1 0 0
00:50.802	ID:1	[ERR : TSCH      ] ! can't send packet to ffff.ffff.ffff.ffff with seqno 166, queue 3 3
01:55.341	ID:1	[ERR : TSCH Queue] ! add packet failed: 0 0x1c82 -1 0 0
01:55.349	ID:1	[ERR : TSCH      ] ! can't send packet to ffff.ffff.ffff.ffff with seqno 166, queue 3 3
02:02.672	ID:2	[ERR : TSCH Queue] ! add packet failed: 0 0x1c82 -1 0 0
02:02.679	ID:2	[ERR : TSCH      ] ! can't send packet to ffff.ffff.ffff.ffff with seqno 28, queue 3 3
02:31.195	ID:2	[ERR : TSCH Queue] ! add packet failed: 0 0x1c82 -1 0 0
02:31.205	ID:2	[ERR : TSCH      ] ! can't send packet to ffff.ffff.ffff.ffff with seqno 28, queue 3 3
02:52.796	ID:2	[ERR : TSCH Queue] ! add packet failed: 0 0x1c82 -1 0 0
02:52.804	ID:2	[ERR : TSCH      ] ! can't send packet to ffff.ffff.ffff.ffff with seqno 28, queue 3 3
03:25.328	ID:2	[ERR : TSCH Queue] ! add packet failed: 0 0x1c82 -1 0 0
03:25.335	ID:2	[ERR : TSCH      ] ! can't send packet to ffff.ffff.ffff.ffff with seqno 29, queue 3 3
03:41.507	ID:2	[ERR : TSCH Queue] ! add packet failed: 0 0x1c82 -1 0 0
03:41.515	ID:2	[ERR : TSCH      ] ! can't send packet to ffff.ffff.ffff.ffff with seqno 29, queue 3 3
04:11.390	ID:2	[ERR : TSCH Queue] ! add packet failed: 0 0x1c82 -1 0 0
04:11.397	ID:2	[ERR : TSCH      ] ! can't send packet to ffff.ffff.ffff.ffff with seqno 29, queue 3 3
04:16.505	ID:1	[ERR : TSCH Queue] ! add packet failed: 0 0x1c82 -1 0 0
04:16.513	ID:1	[ERR : TSCH      ] ! can't send packet to ffff.ffff.ffff.ffff with seqno 166, queue 3 3

The TSCH sched erros are repeated at the begining but I deleted repeated erros to make shorter the error list, the other errors of node 1 and node 2 are periodic. As I said, I was tracking this errors but I did not succeed to find out what is happening.

I will be very grateful if someone could help me with this problem. Basically what I need is to send UDP messages over TSCH orchestra. Maybe there is antoher example which I can base on.

Thanks in advance (sorry about my english).
Martín

atiselsts, 2019-12-02:
You are running out of memory space for the TSCH schedule. Calculate how many TSCH links (active cells) you need to and set that in the config in `project-conf.h`, e.g. if you need 64 links:
`#define TSCH_SCHEDULE_CONF_MAX_LINKS 64`

The timesync-demo is apparently accidentally disabled for z1, it will be fixed in the future - thanks for reporting this.

The Cooja special EB rate also will be fixed in the future.

martincacereal, 2019-12-02:
Dear Atis

Thank you very much for you reply. I tried your advice but still I am stucked at the same point. I am getting the same log. To ilustrate more I am trying with a linear topology, N1 <-> N2<-> N3<-> N4<->N5. only the middle node can see two nodes ande the side node the closet node. The node 1 is de coordinator and root.

I am quite confused what's happening because I would thing this should be working stright away. I am only selecting orchestra from the makefile and commenting de #if line. I am coping tw minutes log of what I am getting to ilustrate this case.

00:01.103	ID:2	[INFO: Main      ] Starting Contiki-NG
00:01.106	ID:2	[INFO: Main      ] - Routing: RPL Lite
00:01.109	ID:2	[INFO: Main      ] - Net: sicslowpan
00:01.111	ID:2	[INFO: Main      ] - MAC: TSCH
00:01.114	ID:2	[INFO: Main      ] - 802.15.4 PANID: 0x81a5
00:01.119	ID:2	[INFO: Main      ] - 802.15.4 TSCH default hopping sequence length: 4
00:01.121	ID:2	[INFO: Main      ] Node ID: 2
00:01.126	ID:2	[INFO: Main      ] Link-layer address: c10c.0000.0000.0002
00:01.134	ID:2	[INFO: Main      ] Tentative link-local IPv6 address: fe80::c30c:0:0:2
00:01.137	ID:2	[INFO: Z1        ] CC2420 CCA threshold -45
00:01.141	ID:2	Orchestra: initializing rule EB per time source (0)
00:01.146	ID:2	Orchestra: initializing rule unicast per neighbor non-storing (1)
00:01.150	ID:2	[ERR : TSCH Sched] ! add_link memb_alloc failed
00:01.154	ID:2	[ERR : TSCH Sched] ! add_link memb_alloc failed
00:01.157	ID:2	[ERR : TSCH Sched] ! add_link memb_alloc failed
00:01.161	ID:2	[ERR : TSCH Sched] ! add_link memb_alloc failed
00:01.164	ID:2	[ERR : TSCH Sched] ! add_link memb_alloc failed
00:01.168	ID:2	[ERR : TSCH Sched] ! add_link memb_alloc failed
00:01.171	ID:2	[ERR : TSCH Sched] ! add_link memb_alloc failed
00:01.174	ID:2	[ERR : TSCH Sched] ! add_link memb_alloc failed
00:01.178	ID:2	[ERR : TSCH Sched] ! add_link memb_alloc failed
00:01.181	ID:2	[ERR : TSCH Sched] ! add_link memb_alloc failed
00:01.185	ID:2	[ERR : TSCH Sched] ! add_link memb_alloc failed
00:01.188	ID:2	[ERR : TSCH Sched] ! add_link memb_alloc failed
00:01.192	ID:2	[ERR : TSCH Sched] ! add_link memb_alloc failed
00:01.195	ID:2	Orchestra: initializing rule default common (2)
00:01.199	ID:2	[ERR : TSCH Sched] ! add_link memb_alloc failed
00:01.201	ID:2	Orchestra: initialization done
00:01.216	ID:4	[INFO: Main      ] Starting Contiki-NG
00:01.218	ID:4	[INFO: Main      ] - Routing: RPL Lite
00:01.221	ID:4	[INFO: Main      ] - Net: sicslowpan
00:01.223	ID:4	[INFO: Main      ] - MAC: TSCH
00:01.227	ID:4	[INFO: Main      ] - 802.15.4 PANID: 0x81a5
00:01.232	ID:4	[INFO: Main      ] - 802.15.4 TSCH default hopping sequence length: 4
00:01.234	ID:4	[INFO: Main      ] Node ID: 4
00:01.238	ID:4	[INFO: Main      ] Link-layer address: c10c.0000.0000.0004
00:01.246	ID:4	[INFO: Main      ] Tentative link-local IPv6 address: fe80::c30c:0:0:4
00:01.249	ID:1	[INFO: Main      ] Starting Contiki-NG
00:01.250	ID:4	[INFO: Z1        ] CC2420 CCA threshold -45
00:01.252	ID:1	[INFO: Main      ] - Routing: RPL Lite
00:01.253	ID:4	Orchestra: initializing rule EB per time source (0)
00:01.255	ID:1	[INFO: Main      ] - Net: sicslowpan
00:01.257	ID:1	[INFO: Main      ] - MAC: TSCH
00:01.258	ID:4	Orchestra: initializing rule unicast per neighbor non-storing (1)
00:01.260	ID:1	[INFO: Main      ] - 802.15.4 PANID: 0x81a5
00:01.263	ID:4	[ERR : TSCH Sched] ! add_link memb_alloc failed
00:01.265	ID:1	[INFO: Main      ] - 802.15.4 TSCH default hopping sequence length: 4
00:01.266	ID:4	[ERR : TSCH Sched] ! add_link memb_alloc failed
00:01.268	ID:1	[INFO: Main      ] Node ID: 1
00:01.269	ID:4	[ERR : TSCH Sched] ! add_link memb_alloc failed
00:01.272	ID:1	[INFO: Main      ] Link-layer address: c10c.0000.0000.0001
00:01.273	ID:4	[ERR : TSCH Sched] ! add_link memb_alloc failed
00:01.276	ID:4	[ERR : TSCH Sched] ! add_link memb_alloc failed
00:01.280	ID:4	[ERR : TSCH Sched] ! add_link memb_alloc failed
00:01.280	ID:1	[INFO: Main      ] Tentative link-local IPv6 address: fe80::c30c:0:0:1
00:01.283	ID:4	[ERR : TSCH Sched] ! add_link memb_alloc failed
00:01.283	ID:1	[INFO: Z1        ] CC2420 CCA threshold -45
00:01.287	ID:4	[ERR : TSCH Sched] ! add_link memb_alloc failed
00:01.287	ID:1	Orchestra: initializing rule EB per time source (0)
00:01.290	ID:4	[ERR : TSCH Sched] ! add_link memb_alloc failed
00:01.292	ID:1	Orchestra: initializing rule unicast per neighbor non-storing (1)
00:01.294	ID:4	[ERR : TSCH Sched] ! add_link memb_alloc failed
00:01.296	ID:1	[ERR : TSCH Sched] ! add_link memb_alloc failed
00:01.297	ID:4	[ERR : TSCH Sched] ! add_link memb_alloc failed
[... 119 lines cut ...]
01:49.851	ID:3	[WARN: 6LoWPAN   ] output: failed to calculate payload size - dropping packet
01:50.879	ID:1	[WARN: 6LoWPAN   ] input: empty packet
01:51.562	ID:4	[INFO: App       ] Not reachable yet
01:52.159	ID:4	[WARN: 6LoWPAN   ] output: failed to calculate payload size - dropping packet
01:53.057	ID:5	[WARN: 6LoWPAN   ] output: failed to calculate payload size - dropping packet
01:55.357	ID:1	[ERR : TSCH Queue] ! add packet failed: 0 0x1c82 -1 0 0
01:55.364	ID:1	[ERR : TSCH      ] ! can't send packet to ffff.ffff.ffff.ffff with seqno 166, queue 3 3
01:56.809	ID:3	[INFO: App       ] Not reachable yet
02:01.854	ID:1	[WARN: 6LoWPAN   ] input: empty packet
02:08.148	ID:3	[WARN: 6LoWPAN   ] output: failed to calculate payload size - dropping packet
02:08.367	ID:5	[INFO: App       ] Not reachable yet
02:09.215	ID:2	[INFO: App       ] Not reachable yet
[... cut ...]
02:10.241	ID:2	[ERR : TSCH      ] ! can't send packet to ffff.ffff.ffff.ffff with seqno 28, queue 3 3

This last messages are repetead indefinitely. What I can see in time line you can see in the attached image. You can see the node 2 get enganged but nothing else happens.

[image]

It is quite a mistery for mw what is going on, but I understand that this example you be running stright away only with orchestra only setting MAKE_WITH_OCHESTRA . Maybe I could try with another plataform, Could you recommend me another one?

Again, thanks a lot for you reply and for your time.
Cherrs!

Martín

atiselsts, 2019-12-03:
Hi Martin,

Please follow my recommendation and set the `TSCH_SCHEDULE_CONF_MAX_LINKS` to a sufficient number in the `project-conf.h` - I just verified that it helps. The z1 platform by default hasa reduced number of TSCH links to save RAM. You are trying to do something that is not supported out of the box in the OS. The example was clearly not tested with Orchestra on z1 platform, as the z1 platform was disabled in the Makefile.

Yes the example would work out of the box with Cooja motes and Orchestra.

yatch, 2020-01-09:
@martincacereal did you resolve the issue with the recommendation of https://github.com/contiki-ng/contiki-ng/issues/1146#issuecomment-561047560 ? I'm closing the issue now; feel free to reopen this if the issue remains.

closed 2020-01-09 by yatch