contiki-ng/contiki-ng #1152: Low Power mode for CC1310

NeoN7777, 2019-12-11:
Hello!
What is the right way to use low power mode for contiki-ng with TSCH? 
@StefanHri asked the same question, but there is no answers https://github.com/contiki-ng/contiki-ng/issues/907
From contiki wiki:
When there are no events in the Contiki-NG event queue, the chip will enter 'some' low power mode (more below).
In "hello world" example with MAKE_MAC = MAKE_MAC_NULLMAC MAKE_NET = MAKE_NET_NULLNET defines in makefile power consumption is about 7 mA. What is the proper consumption for cc13x0 with TSCH and how I can enter to low poer mode? Thanks!

NeoN7777, 2019-12-12:
Maybe anybody can suggest TSCH configs to decrease power consumption in 6lowpan?

atiselsts, 2019-12-12:
Hi,

Disable the default Orchestra non-storing rule, as it wakes up the system in every TSCH slot. Some platforms might even fail to deep sleep altogether.

To do that, configure the storing mode rule in `project-conf.h`:

    #define ORCHESTRA_CONF_RULES { &eb_per_time_source, &unicast_per_neighbor_rpl_storing, &default_common }

You also need to switch to RPL Classic in the `Makefile`, as RPL Lite does not support the storing mode:
   
    MAKE_ROUTING = MAKE_ROUTING_RPL_CLASSIC

NeoN7777, 2019-12-13:
> Hi,
>
> Disable the default Orchestra non-storing rule, as it wakes up the system in every TSCH slot. Some platforms might even fail to deep sleep altogether.
>
> To do that, configure the storing mode [...]
Hi, atiselsts, thanks for reply!
Power consuption is still about 150 uA, is it ok for TSCH?

chenek, 2019-12-25:
I think 150 uA power consumption is ok for TSCH.

closed 2020-10-22 by atiselsts