contiki-ng/contiki-ng #1154: very_sleepy_demo removes default IPv6 Route

ahmedmibrahim, 2019-12-12:
- Running very_sleepy_Demo on cc1350 Launchpad with Native Border Router on a RaspberryPi
- Using RPL_Lite, I noticed the very_sleepy_node removing the default IPv6 route every while which causes it to search again for the same neighbour and stays awake longer.
- When I send a CoAP request during the nbr=NULL period, the node recognizes the request but doesn't respond (I found that from LOG_LEVEL_DBG of Coap-engine) which means it is still connected to the network

Please see the LOG_DBG of IPV6 below:

 DBG: Inside switch_to_very_sleepy
[INFO: IPv6 Route] Default route, IP address fe80::212:4b00:1375:b481
[INFO: IPv6 Route] Default route found, IP address fe80::212:4b00:1375:b481
 DBG: Inside switch_to_normal
[INFO: IPv6 Route] Default route, IP address fe80::212:4b00:1375:b481
[INFO: IPv6 Route] Default route found, IP address fe80::212:4b00:1375:b481
 DBG: switch_to_very_sleepy
[INFO: IPv6 Route] Default route, IP address fe80::212:4b00:1375:b481
[INFO: IPv6 Route] Default route found, IP address fe80::212:4b00:1375:b481
 DBG: Inside switch_to_normal
[INFO: IPv6 Route] Default route, IP address fe80::212:4b00:1375:b481
[INFO: IPv6 Route] Default route found, IP address fe80::212:4b00:1375:b481
 DBG: Inside switch_to_very_sleepy
[INFO: IPv6 Route] Removing default           // No Idea why this happened!
        DBG: Inside keep_mac_on() --> (nbr = uip_ds6_nbr_lookup(uip_ds6_defrt_choose()); returned NULL)
DBG: Inside switch_to_normal
        DBG: nbr == NULL
DBG: Inside switch_to_very_sleepy
        DBG: nbr == NULL
        DBG: nbr == NULL

closed 2019-12-12 by ahmedmibrahim

DaSchaef, 2019-12-16:
Could you explain why this is closed? I'm having problems with deleted default route too, leading to pretty high number of connectivity loss.

ahmedmibrahim, 2019-12-16:
Because you need to keep mac on for at least 10 seconds (this is explained in a comment just above the keep_mac_on function.

It actually makes it a bad demo as you will lose more battery power if you have to keep the radio on for 10 complete seconds.

DaSchaef, 2019-12-17:
Oh no, I looked so long in the code and didn't see it. Thank you!! :-)
But I found another thing, maybe helping you (I'm totally new to IOT and Contiki, even though).

In the seriell log, I see that my default route is always lost if a single DAO Ack was lost.
In the code in  rpl-timer.c, there is also a commet about /* Send between 60 and 120 seconds before target refresh */.

I set RPL_WITH_DAO_ACK to 0 in my local project-conf.h.
And from rpl-conf.h there is a comment to then increase
RPL_TRICKLE_REFRESH_DAO_ROUTES  for the RPL root (border router in my case).

This seem to make situation a lot better.
Packet loss is down to <2% from 20%-30%.

Not looked at power yet.

rpl-conf.h
```/*
 * Setting the RPL_TRICKLE_REFRESH_DAO_ROUTES will make the RPL root
 * increase the DTSN (Destination Advertisement Trigger Sequence Number)
 * from the DIO trickle timer. If set to 4, DTSN will be increased every 4th
 * iteration. This is to get all children to re-register their DAO route.
 * This is needed when DAO-ACK is not enabled to
 * add reliability to route maintenance.
 * */```

rpl-timer.c
```schedule_dao_refresh(void)
{
  if(curr_instance.used && curr_instance.default_lifetime != RPL_INFINITE_LIFETIME) {
#if RPL_WITH_DAO_ACK
    /* DAO-ACK enabled: the last DAO was ACKed, wait until expiration before refresh */
    clock_time_t target_refresh = CLOCK_SECOND * RPL_LIFETIME(curr_instance.default_lifetime);
#else /* RPL_WITH_DAO_ACK */
    /* DAO-ACK disabled: use half the expiration time to get two chances to refresh per lifetime */
    clock_time_t target_refresh = (CLOCK_SECOND * RPL_LIFETIME(curr_instance.default_lifetime) / 2);
#endif /* RPL_WITH_DAO_ACK */

    /* Send between 60 and 120 seconds before target refresh */
    clock_time_t safety_margin = (60 * CLOCK_SECOND) + (random_rand() % (60 * CLOCK_SECOND));

    if(target_refresh > safety_margin) {
      target_refresh -= safety_margin;
    }

    /* Schedule transmission */
    ctimer_set(&curr_instance.dag.dao_timer, target_refresh, send_new_dao, NULL);
  }
}
```

ahmedmibrahim, 2019-12-19:
Thanks can you please list your timing details with these new things you mentioned? 

> /*---------------------------------------------------------------------------*/
/* Normal mode duration params in seconds */
#define NORMAL_OP_DURATION_DEFAULT 10    //10
#define NORMAL_OP_DURATION_MIN     10    //10
#define NORMAL_OP_DURATION_MAX     60    //60
/*---------------------------------------------------------------------------*/
/* Observer notification period params in seconds */
#define PERIODIC_INTERVAL_DEFAULT  20     //30
#define PERIODIC_INTERVAL_MIN      20     //30
#define PERIODIC_INTERVAL_MAX      86400  //86400 /* 1 day */

I'm not an expert in RPL at all, but based on the comment above keep_mac_on, I'd say we just need to get the NBR to send the NA (I assume this is Neighbour Acknowledge) faster so that we can keep the mac on for only a fraction of a second or so.
Did you figure out how to control this NA sending?

DaSchaef, 2019-12-19:
I changed them a bit (10sek alive, 2min offtime), but my main change is:
````
/* Schedule next pass */
      if(mac_status) {
        etimer_set(&et_periodic, PERIODIC_INTERVAL_ON); // 1s
      } else {
        etimer_set(&et_periodic, PERIODIC_INTERVAL_OFF); // 10s
      }
````

mac_status is set depending on NETSTACK_MAC.on(); NETSTACK_MAC.off();

It is still not optimal, and I'm having same short 4min connection losses each day. But this averages to approx < 2 % packet loss.

ahmedmibrahim, 2019-12-21:
with 10 seconds ON and 60 minutes sleeping, I ran a 1 day test and here are my results for coin-cell (225mAh and 1Ah) with battery factor of 0.7:

  | LPM | Deep LPM | CPU | RADIO Tx | RADIO Rx
-- | -- | -- | -- | -- | --
Duration percentage | 4.457% | 95.374% | 0.169% | 0.042% | 4.441%
current (mA) | 6.646 | 0.01 | 7 | 13.4 | 5.4

average current  (mA)   | 0.563031
-- | --
Days of operation (225mAh)  | 11.65565
Days of operation (1Ah)   | 51.80291

Still not enough for a real application ....

chenek, 2019-12-25:
I don't see any power consumption could be improved with 10 seconds ON and 10 minutes sleeping.