contiki-ng/contiki-ng #1175: Failed to build examples for target jn516x

hlukasz, 2020-01-16:
Tried hello-world and platform specific example, both develop and release/v4.4 branches.
Error during linking:

```
/opt/ba-elf-gcc/bin/../lib/gcc/ba-elf/4.7.4/../../../../ba-elf/lib/mcpu_jn51xx/optsize/libc.a(lib_a-glue.o): In function `_sbrk':
/media/secondary/r36379/gcc-4.7.4-ba-r36379-build/ba-elf/mcpu_jn51xx/optsize/newlib/libc/sys/basim/../../../../../../../../gcc-4.7.4-ba-r36379/newlib/libc/sys/basim/glue.c:75: undefined reference to `end'
```
Full log: [log.txt](https://github.com/contiki-ng/contiki-ng/files/4073688/log.txt)

```
$ ba-elf-gcc -v                                                                                                                                                                                                            
Using built-in specs.
COLLECT_GCC=ba-elf-gcc
COLLECT_LTO_WRAPPER=/opt/ba-elf-gcc/bin/../libexec/gcc/ba-elf/4.7.4/lto-wrapper
Target: ba-elf
Configured with: ../gcc-4.7.4-ba-r36379/configure --host= --target=ba-elf --prefix=/usr/jn-toolchain/ --enable-languages=c,c++,lto --with-gnu-as --with-gnu-ld --with-newlib --enable-target-optspace --disable-libssp --disable-__cxa_atexit --with-gxx-include-dir=/usr/jn-toolchain//ba-elf/include
Thread model: single
gcc version 4.7.4 (GCC)
```
```
$ cat /usr/jn516x-sdk/JN-SW-4163/build.txt
Build Number 1416
```

With old contiki (non-ng) build is passing.

yatch, 2020-01-17:
`examples/hello-world` is built without any error under [the docker environment (contiker)](https://github.com/contiki-ng/contiki-ng/wiki/Docker). I tried on the latest `develop` branch:

```
$ git rev-parse --short HEAD
b991985
$ pwd
/home/user/contiki-ng/examples/hello-world
$ make TARGET=jn516x
  MKDIR     build/jn516x/obj
  CC        ../../arch/platform/jn516x/dev/jn516x-ccm-star.c
  CC        ../../arch/platform/jn516x/dev/exceptions.c
(snip)
  CC        ../../arch/platform/jn516x/./platform.c
Creating empty build/jn516x/hello-world.jn516x
touch build/jn516x/hello-world.jn516x
  CP        build/jn516x/hello-world.jn516x --> hello-world.jn516x
rm hello-world.o build/jn516x/obj/platform.o

$ ba-elf-gcc -v
Using built-in specs.
COLLECT_GCC=ba-elf-gcc
COLLECT_LTO_WRAPPER=/usr/ba-elf-gcc/bin/../libexec/gcc/ba-elf/4.7.4/lto-wrapper
Target: ba-elf
Configured with: ../gcc-4.7.4-ba-r36379/configure --host= --target=ba-elf --prefix=/usr/jn-toolchain/ --enable-languages=c,c++,lto --with-gnu-as --with-gnu-ld --with-newlib --enable-target-optspace --disable-libssp --disable-__cxa_atexit --with-gxx-include-dir=/usr/jn-toolchain//ba-elf/include
Thread model: single
gcc version 4.7.4 (GCC)
$ cat /usr/jn516x-sdk/JN-SW-4163/build.txt
Build Number 1416
```
I guess, your (local?) environment is not properly configured for the build. The following wiki page may help:

* https://github.com/contiki-ng/contiki-ng/wiki/Platform-jn516x

If you followed the wiki page, the contents may be incorrect. It'd be great if you could point out what is missing or wrong.

Or, you might want to take a look at `tools/docker/Dockerfile` to see how the docker image is configured.

hlukasz, 2020-01-17:
It looks like that the problem is connected with SDK (JN-SW-4163) downloaded from NXP site. After replacing it with one downloaded from site listed in Dockerfile (simonduq.github.io), everything is working. There are some minor differences between these two versions.

Anyway, it may be worth to mention in wiki that recommended source for SDK is from simonduq.github.io

hlukasz, 2020-01-17:
I claim victory too fast. Build is passing, but result file has 0B...
@yatch In your logs also empty file was created: "Creating empty build/jn516x/hello-world.jn516x"
After looking at Makefile.jn516x this happens when SDK has no libraries, that indeed is true when using SDK from Dockerfile.

yatch, 2020-01-18:
Make sure to follow this instruction https://github.com/contiki-ng/contiki-ng/wiki/Platform-jn516x#install-the-toolchain ? You need to have the SDK fully installed. In other words, `/usr/jn516x-sdk/JN-SW-4163/Components/Library/` shouldn't empty.

hlukasz, 2020-01-18:
I installed docker and tried with contiker. Inside docker image `/usr/jn516x-sdk/JN-SW-4163/Components/Library/` is empty and hello-world binary has 0 bytes after build. Could you please check how does it look in your setup? I have no idea what can I do wrong, docker setup procedure is very clear.

I followed docker guide from wiki: https://github.com/contiki-ng/contiki-ng/wiki/Docker

yatch, 2020-01-19:
The docker image doesn't contain the SDK.

You need to download the SDK through the web form, anyway. Then,

* use the SDK with the toolchain installed (natively) on your local machine

or

* mount a directory having the SDK from a docker container, and use it the toolchain with the toolchain in the docker image.

hlukasz, 2020-01-19:
I was trying to mount SDK downloaded from NXP, but that leads to "undefined reference to `end'". I am quite sure that this problem is not related with my environment. Moreover, with v4.1 build is passing using exactly the same environment (SDK from NXP) both inside Docker and on my host. It looks like something is wrong starting from v4.2

yatch, 2020-01-22:
@hlukasz can you give logs by the following commands to get helps from others?

```
[under the environment where v4.1 can be built]
$ cd examples/hello-world
$ git checkout release/v4.1 
$ make TARGET=jn516x clean
$ make TARGET=jn516x

$ git checkout release/v4.2
$ make TARGET=jn516x clean
$ make TARGET=jn516x
```

In addition, it would be nice if you can give what you got in the docker container where the SDK was mounted under `/usr/jn516x-sdk/JN-SW-4163` during the compilation.

Thanks!

hlukasz, 2020-01-22:
Ok, here are logs, everything was done inside docker, same environment, same project.
[build_v4.1.txt](https://github.com/contiki-ng/contiki-ng/files/4095953/build_v4.1.txt)
[build_v4.2.txt](https://github.com/contiki-ng/contiki-ng/files/4095955/build_v4.2.txt)

Content of `/usr/jn516x-sdk`: [sdk_tree.txt](https://github.com/contiki-ng/contiki-ng/files/4095959/sdk_tree.txt)

I did an experiment and add `. = end;` in `JN-SW-4163/Chip/JN5168/Build/AppBuildEnd.ld` at the end of `.bss` section, build was successful but CPU didn't start at all after flashing.

yatch, 2020-01-22:
Great. I will open a new issue later with the logs you provided 👍

yatch, 2020-01-22:
Continue on #1178

closed 2020-01-22 by yatch