contiki-ng/contiki-ng #1177: Displaying DODAG Tree

fkerem, 2020-01-18:
In the previous versions of Contiki, I was able to display the DODAG tree by adding following line to uip-ds6-route.c file:
#include DEBUG DEBUG_ANNOTATE

However, I couldn't figure it out in contigi-ng.

Can anyone help me?

yatch, 2020-01-19:
Try adding the following line into your `project-conf.h`:

```
#define LOG_CONF_WITH_ANNOTATE 1
```

I confirmed this works with `examples/rpl-udp`.

[image]

fkerem, 2020-01-19:
But which project-conf.h you're talking about?
There are many project-conf.h:

./tests/13-ieee802154/code-flush-nbr-queue/project-conf.h
./tests/13-ieee802154/code-6tisch/project-conf.h
./tests/13-ieee802154/code-panid-handling/project-conf.h
./tests/07-simulation-base/code-slip-radio/project-conf.h
./tests/07-simulation-base/code-ipv6/sender/project-conf.h
./tests/07-simulation-base/code-ipv6/receiver/project-conf.h
./tests/07-simulation-base/code-data-structures/project-conf.h
./tests/07-simulation-base/code-ringbufindex/project-conf.h
./tests/14-rpl-lite/code/project-conf.h
./tests/15-rpl-classic/code/project-conf.h
./examples/lwm2m-ipso-objects/project-conf.h
./examples/websocket/project-conf.h
./examples/slip-radio/project-conf.h
./examples/multicast/project-conf.h
./examples/rpl-border-router/project-conf.h
./examples/storage/cfs-coffee/project-conf.h
./examples/6tisch/etsi-plugtest-2017/project-conf.h
./examples/6tisch/simple-node/project-conf.h
./examples/6tisch/timesync-demo/project-conf.h
./examples/6tisch/channel-selection-demo/project-conf.h
./examples/6tisch/tsch-stats/project-conf.h
./examples/6tisch/sixtop/project-conf.h
./examples/6tisch/6p-packet/project-conf.h
./examples/libs/energest/project-conf.h
./examples/libs/stack-check/project-conf.h
./examples/libs/deployment/project-conf.h
./examples/libs/ipv6-uipbuf/project-conf.h
./examples/libs/logging/project-conf.h
./examples/platform-specific/nrf52dk/mqtt-demo/project-conf.h
./examples/platform-specific/nrf52dk/coap-demo/project-conf.h
./examples/platform-specific/cc26x0-cc13x0/very-sleepy-demo/project-conf.h
./examples/platform-specific/cc26x0-cc13x0/ble-ipv6/project-conf.h
./examples/platform-specific/cc26x0-cc13x0/cc26x0-web-demo/project-conf.h
./examples/platform-specific/cc26x0-cc13x0/project-conf.h
./examples/platform-specific/zoul/at-test/project-conf.h
./examples/platform-specific/zoul/orion/ip64-router/project-conf.h
./examples/platform-specific/zoul/orion/client/project-conf.h
./examples/platform-specific/zoul/project-conf.h
./examples/platform-specific/zoul/rtcc/project-conf.h
./examples/platform-specific/zoul/rev-b/project-conf.h
./examples/platform-specific/jn516x/rpl/coap-dr1199-node/project-conf.h
./examples/platform-specific/jn516x/rpl/coap-dr1175-node/project-conf.h
./examples/platform-specific/jn516x/rpl/coap-dongle-node/project-conf.h
./examples/platform-specific/jn516x/rpl/node/project-conf.h
./examples/platform-specific/jn516x/tsch/simple-sensor-network/node/project-conf.h
./examples/platform-specific/jn516x/tsch/tx-power-verification/node/project-conf.h
./examples/platform-specific/jn516x/tsch/uart1-test-node/project-conf.h
./examples/sensniff/project-conf.h
./examples/ip64-router/project-conf.h
./examples/benchmarks/rpl-req-resp/project-conf.h
./examples/dev/button-hal/project-conf.h
./examples/coap/coap-example-server/project-conf.h
./examples/coap/coap-example-client/project-conf.h
./examples/coap/coap-plugtest-server/project-conf.h
./examples/mqtt-client/project-conf.h
./examples/snmp-server/project-conf.h

fkerem, 2020-01-19:
Since I don't have a project-conf.h inside ./examples/rpl-udp, I've added one and added the line you stated. However, I got such a compilation error during the compilation of udp server:

[image]

fkerem, 2020-01-19:
It works pretty well when I'm using RPL-Lite but it results in a compilation error in RPL-Classic:

[image]

fkerem, 2020-01-20:
I think the problem in _**uip-ds6-route.c**_ file occurs here but I couldn't figure it out:

[image]

yatch, 2020-01-22:
@fkerem Thanks for reporting the issue; I'm working on it.

yatch, 2020-01-30:
@fkerem Sorry for taking so long. I had troubles on my machine... Try putting `const` before `uip_ipaddr_t`; it should work. I will open a PR for this.

[image]

fkerem, 2020-01-30:
> @fkerem Sorry for taking so long. I had troubles on my machine... Try putting `const` before `uip_ipaddr_t`; it should work. I will open a PR for this.
> 
> [image]

I really appreciate your help! Thank you so much!

yatch, 2020-02-05:
Closed via PR #1185

closed 2020-02-05 by yatch

yasben11, 2020-02-20:
Hello evryone
I'm trying to run the rpl-udp-sky.csc example (contiki-ng/exemples/rpl-udp), I want to display the dodag tree.
can somone help me to fix that.
thank you

yatch, 2020-02-22:
@yasben11 open a new file at `rpl-udp` directory, put the following line, save it as `project-conf.h`, then start a new simulation:

```
#define LOG_CONF_WITH_ANNOTATE 1
```