contiki-ng/contiki-ng #1178: Build for TARGET=jn516x fails with v4.2

yatch, 2020-01-22:
For instance, `examples/hello-world` fails with the following error:

```
ba-elf-gcc -Wl,--gc-sections -TApp_Stack_Size.ld -Wl,--fatal-warnings -march=ba2 -mcpu=jn51xx -mredzone-size=4 -mbranch-cost=3 -fomit-frame-pointer -Os -fshort-enums -nostartfiles -L/usr/jn516x-sdk/JN-SW-4163/Chip/JN5168/Build -L/usr/jn516x-sdk/JN-SW-4163/Chip/JN5168/Library -L/usr/jn516x-sdk/JN-SW-4163/Platform/DK4/Library -L/usr/jn516x-sdk/JN-SW-4163/Components/Library -L/usr/jn516x-sdk/JN-SW-4163/Stack/MAC/Build -L/usr/jn516x-sdk/JN-SW-4163/Chip/JN5168/Build -L/usr/jn516x-sdk/JN-SW-4163/Chip/JN5168/Library -Wl,--defsym=_stack=_stack_low_water_mark -Wl,--defsym=_stack_origin=_ram_top -TAppBuildNone.ld -o build/jn516x///hello-world.jn516x -Wl,--start-group \
  hello-world.o build/jn516x///obj/platform.o \
  build/jn516x///contiki-ng-jn516x.a \
  -lMMAC_JN516x -lAes_JN516x -lHardwareApi_JN516x -lMicroSpecific_JN516x -lBoot_JN516x -lPDM_EEPROM_JN516x_NO_RTOS -lBoardLib_JN516x -lRecal_JN516x -lJPT_JN5168   -Wl,--end-group -Wl,-Map,build/jn516x///hello-world.map
/usr/ba-elf-gcc/bin/../lib/gcc/ba-elf/4.7.4/../../../../ba-elf/lib/mcpu_jn51xx/optsize/libc.a(lib_a-glue.o): In function `_sbrk':
/media/secondary/r36379/gcc-4.7.4-ba-r36379-build/ba-elf/mcpu_jn51xx/optsize/newlib/libc/sys/basim/../../../../../../../../gcc-4.7.4-ba-r36379/newlib/libc/sys/basim/glue.c:75: undefined reference to `end'
collect2: error: ld returned 1 exit status
../../arch/platform/jn516x/Makefile.jn516x:234: recipe for target 'build/jn516x///hello-world.jn516x' failed
make: *** [build/jn516x///hello-world.jn516x] Error 1
rm hello-world.o build/jn516x///obj/platform.o
```

Under the same environment, `hello-world` in v4.1 is built without errors:
```
ba-elf-gcc -Wl,--gc-sections -TApp_Stack_Size.ld -march=ba2 -mcpu=jn51xx -mredzone-size=4 -mbranch-cost=3 -fomit-frame-pointer -Os -fshort-enums -nostartfiles -L/usr/jn516x-sdk/JN-SW-4163/Chip/JN5168/Build -L/usr/jn516x-sdk/JN-SW-4163/Chip/JN5168/Library -L/usr/jn516x-sdk/JN-SW-4163/Platform/DK4/Library -L/usr/jn516x-sdk/JN-SW-4163/Components/Library -L/usr/jn516x-sdk/JN-SW-4163/Stack/MAC/Build -L/usr/jn516x-sdk/JN-SW-4163/Chip/JN5168/Build -L/usr/jn516x-sdk/JN-SW-4163/Chip/JN5168/Library -Wl,--defsym=_stack=_stack_low_water_mark -Wl,--defsym=_stack_origin=_ram_top -TAppBuildNone.ld -o hello-world.jn516x -Wl,--start-group \
  hello-world.o obj_jn516x/platform.o \
  contiki-ng-jn516x.a \
  -lMMAC_JN516x -lAes_JN516x -lHardwareApi_JN516x -lMicroSpecific_JN516x -lBoot_JN516x -lPDM_EEPROM_JN516x_NO_RTOS -lBoardLib_JN516x -lRecal_JN516x -lJPT_JN5168   -Wl,--end-group -Wl,-Map,hello-world-jn516x.map
ba-elf-objcopy -S -O binary hello-world.jn516x hello-world.jn516x.bin
rm obj_jn516x/platform.o hello-world.o
user@8ce009348d55:~/contiki-ng/examples/hello-world$ ls -lh hello-world.jn516x
-rwxr-xr-x 1 user user 119K Jan 22 08:20 hello-world.jn516x
```

Full logs and the view of the SDK directory are found in the attached files:

* [build_v4.2.txt](https://github.com/contiki-ng/contiki-ng/files/4096195/build_v4.2.txt)
* [Uploading build_v4.1.txt…]()
* [sdk_tree.txt](https://github.com/contiki-ng/contiki-ng/files/4096203/sdk_tree.txt)

This issue was reported by @hlukasz, who provided all the files mentioned above. The original issue ticket is Issue #1175.

g-oikonomou, 2020-01-22:
Possibly some of the build system changes. I'd start looking with #785 and #688. Less likely #698, #865

hlukasz, 2020-01-23:
@g-oikonomou You are right, reverting changes made in #688 for `Makefile.jn516x` fixes building. Correct me if I am wrong - as JN516x is not an ARM core, so config from `Makefile.arm` is not taken into consideration. That's why you have to leave `dbg-io` in `Makefile.jn516x`.

g-oikonomou, 2020-01-23:
Please open a PR to put the removed line back 😄

hlukasz, 2020-01-24:
Done #1180

g-oikonomou, 2020-01-24:
Fixed in #1180

closed 2020-01-24 by g-oikonomou