contiki-ng/contiki-ng #1182: Unexpected Rtimer behavior

hseokman, 2020-01-25:
From looking at how Rtimer is implemented, it is confusing how this was designed.

The immediate problem is when two rtimers are scheduled. If the second rtimer is scheduled before the first rtimer is executed, then rtimer will run the second rtimer function with the first rtimer timeout period.

It is confusing if this timer was meant to handle multiple timers or just single timer.
It seems it was designed to handle multiple timers since it is checking if the timer is the first timer or not and scheduling the next one is done at the handler, but is is missing code for maintaining the list of timers.

yanpads, 2020-01-27:
Pull request https://github.com/contiki-ng/contiki-ng/pull/556 is also related to two rtimer scheduling.

atiselsts, 2020-01-31:
The existing rtimer implementation only supports a single rtimer.

yanpads, 2020-02-01:
Hi @atiselsts , 

What we mean is `rtimer_arch_schedule` is always called twice. Please refer to the comment https://github.com/contiki-ng/contiki-ng/pull/556#issuecomment-426278994