contiki-ng/contiki-ng #1184: UDP+TSCH - Storing with Orchestra

GaryLu889, 2020-01-31:
I am not sure if this is the right place to ask, but here it goes

I am trying to set up a TSCH with UDP communication, briefly the overall system should operate as below:

1) Root and Node establishes TSCH - synchronising timestamps  
2) Once Node has 'found' or established TSCH communication with root, it sends UDP packet (act as a client) to Root
3) Root then send a reply to Node via UDP

The problem I encountered when combing TSCH and UDP examples is Node not be able to receive the replying message from Root.

Which it is then found out that it has something to do within the project configuration header (RPL non-storing mode), specifically the following : 

#undef RPL_CONF_MOP
#define RPL_CONF_MOP RPL_MOP_NON_STORING /* Mode of operation*/

So I begin to manipulate around with it,  but I cant seem to be understanding the concept correctly. 

Question I have is, what is the main difference between RPL storing and Non-storing? Is there an example of RPL storing method?

On the other hand, TSCH with Storing orchestra, how does it work differently with Non storing Orchestra? 

Thank you in advance for your time.

Regards

yanpads, 2020-01-31:
Regarding your question about difference between RPL storing and non-storing mode, I think the documentation about RPL contains information about it.

https://github.com/contiki-ng/contiki-ng/wiki/Documentation:-RPL

atiselsts, 2020-01-31:
To enable the storing mode, add this to the project's Makefile (it will select RPL Classic).

`MAKE_ROUTING = MAKE_ROUTING_RPL_CLASSIC`

You can also add this to project-conf.h: `#define RPL_CONF_MOP RPL_MOP_STORING_NO_MULTICAST`, which is also the default mode of operation of RPL Classic.

If you believe that you have found a problem, submit a new issue with enough information so that we can reproduce it.

closed 2020-01-31 by atiselsts