contiki-ng/contiki-ng #1190: UnsupportedClassVersionError: org/contikios/cooja/corecomm/Lib1 has been compiled by a more recent version of the Java Runtime (class file version 55.0), this version of the Java Runtime only recognizes class file versions up to 52.0

iliar-phd, 2020-02-11:
Hi,

Trying to compile and create some code in cooja but after a successful compilation, the window for creating the motes does not show up and this error in the terminal:
     [java] Exception in thread "AWT-EventQueue-1" java.lang.UnsupportedClassVersionError: org/contikios/cooja/corecomm/Lib1 has been compiled by a more recent version of the Java Runtime (class file version 55.0), this version of the Java Runtime only recognizes class file versions up to 52.0

I am using Java 1.8.0_242. Java_Home is set correctly on Ubuntu 18.04

iliar-phd, 2020-02-11:
The code I want to run on "cooja" mote is "examples/rpl-udp/udp-client.c"

jbrusey, 2021-11-04:
I also came across this. Somehow two versions of openjdk (8 and 11) were installed. The solution was to:

    sudo update-alternatives --config java

and

    sudo update-alternatives --config javac

setting both to version 8 (for example).

closed 2022-06-22 by joakimeriksson