contiki-ng/contiki-ng #1191: NG Shell not working?

LukasEsterle, 2020-02-13:
Hi,
i am trying to use the shell in contiki-ng and as per example i added 'MODULES += os/services/shell' to the makefile of my project (and the hello-world example makefile).
I am using the vagrant image to upload the image to the node and this SEEMS to work fine, even though i expected some kind of output with respect to the shell during compile-time.

i understand that vagrant might give me issues with the the EOL command. So i also tried coolterm natively on my mac (which supposedly works fine). but nothing. After connecting to the ode (via CoolTerm natively from the mac), i see the outputs of the mote. i can use the 'send string' command and a visual feedback fo CoolTerm tells me it has been transmitted, but no reply from the node. 

given there is no cue on whether the shell has been included in the binary and uploaded to the mote, i was wondering if something went wrong. I also changed the command to 'MODULES += $(CONTIKI)/os/services/shell' but still no success. i obviously used make distclean after each change.

Best,
Lukas

yatch, 2020-02-16:
What platform/board are you using? 

> After connecting to the ode (via CoolTerm natively from the mac), i see the outputs of the mote. i can use the 'send string' command and a visual feedback fo CoolTerm tells me it has been transmitted, but no reply from the node.

Can you paste the output here? Need more details to help.

LukasEsterle, 2020-02-16:
i've got a XBow TelosB and an MTM-CM5000MSP (sky platform).

>>After connecting to the ode (via CoolTerm natively from the mac), i see the outputs of the mote. i can use the 'send string' command and a visual feedback fo CoolTerm tells me it has been transmitted, but no reply from the node.
> Can you paste the output here? Need more details to help.

there is no actual output, CoolTerm only has this small visual feedback (represented as an LED, i guess, that tells me something has been transmitted).

yatch, 2020-02-16:
Did you check the serial settings including baud rate?

LukasEsterle, 2020-02-19:
Yes, I do get output via the serial port from my running program and that is all okay.
i also tried different settings fro the line states and the flow control. 
(I also played around with different Data Bits, Parity and Stop Bits, but no success)

closed 2020-11-25 by g-oikonomou

CodeDragon82, 2023-03-20:
> 

Did you every solve this?

asereq, 2023-08-22:
It is not explained in the [Contiki-NG Shell Tutorial](https://docs.contiki-ng.org/en/master/doc/tutorials/Shell.html), but the line `MODULES += os/services/shell` must be added BEFORE the line `include $(CONTIKI)/Makefile.include`.

I am going to do a pull request about that.