contiki-ng/contiki-ng #1206: Contiki-NG Getting Started | Documentation Page

bitvijays, 2020-03-10:
This could be a [Getting started](https://github.com/contiki-ng/wiki-sandbox/wiki/Contiki-NG-Getting-Started) page

ChalinoHu, 2020-09-20:
This is my first time to use the CCS with the Contiki-ng,I configured the CCS refer to the document[https://github.com/contiki-ng/contiki-ng/wiki/Platform-simplelink#setup-contiki-ng-in-Code-Composer-Studio]( https://github.com/contiki-ng/contiki-ng/wiki/Platform-simplelink#setup-contiki-ng-in-Code-Composer-Studio)
How to configure the CCS by "Set Program to the absolute path of the generated .elf file. It is recommended to click File System and navigate to the .elf file manually to make sure the path is correct."?
[image]

And I'can't build the project with the console "gmake: *** No rule to make target 'hello-world.elf'.	hello-world		 	C/C++ Problem"
[image]

Can someone help me?

arurke, 2024-07-07:
Assuming this is moot after we moved to "Read the Docs" at [docs.contiki.org](https://docs.contiki-ng.org/)

closed 2024-07-07 by arurke