contiki-ng/contiki-ng #1208: Missing parentheses on define

nbrunner, 2020-03-12:
in contiki-ng/os/net/mac/csma/csma.h
CSMA_ACK_WAIT_TIME and CSMA_AFTER_ACK_DETECTED_WAIT_TIME
in contiki-ng/arch/platform/cooja/contiki-conf.h
CSMA_CONF_ACK_WAIT_TIME

rajeev1986, 2020-04-13:
@nbrunner 
When using #define, parentheses are not actually required per se, but if your compilation is giving errors, then go ahead and add it.

nbrunner, 2020-04-14:
Lots of whitespace characters are also not actually required, but it's not a reason for not using them.
Currently it's doesn't make a compile or runtime error but it can avoid potential future problem.

atiselsts, 2020-10-22:
We don't have a coding policy in Contiki-NG that requires parentheses around all define statements, so it not strictly an issue.

closed 2020-10-22 by atiselsts